isaac-sim/IsaacGymEnvs #2: Cannot clone repo: "This repository is over its data quota. Account responsible for LFS bandwidth should purchase more data packs to restore access."

JohannLange, 2021-11-05:
I tried cloning the repo, but it seems some objects are stored in the LFS and the bandwidth is all used up.
Whenever I try to clone the repo, first, I get

```text
Cloning into 'IsaacGymEnvs'...
remote: Enumerating objects: 568, done.
remote: Counting objects: 100% (568/568), done.
remote: Compressing objects: 100% (450/450), done.
remote: Total 568 (delta 114), reused 568 (delta 114), pack-reused 0
Receiving objects: 100% (568/568), 25.76 MiB | 9.37 MiB/s, done.
Resolving deltas: 100% (114/114), done.
Downloading assets/trifinger/objects/meshes/ball.obj (5.6 KB)
Error downloading object: assets/trifinger/objects/meshes/ball.obj (88c1693): Smudge error: Error downloading assets/trifinger/objects/meshes/ball.obj (88c16939f216d501a06685b476edf01e8e41f01c4124882549186534c6cee39d): batch response: Post https://github.com/JohannLange/IsaacGymEnvs.git/info/lfs/objects/batch: x509: certificate signed by unknown authority

Errors logged to /home/johannlange/tmp/IsaacGymEnvs/.git/lfs/logs/20211105T151111.3103445.log
Use `git lfs logs last` to view the log.
error: external filter 'git-lfs filter-process' failed
fatal: assets/trifinger/objects/meshes/ball.obj: smudge filter lfs failed
warning: Clone succeeded, but checkout failed.
You can inspect what was checked out with 'git status'
and retry with 'git restore --source=HEAD :/'
```

Afterwards, when I try to manually fetch the LFS objects, I get

```text
fetch: Fetching reference refs/heads/main
batch response: This repository is over its data quota. Account responsible for LFS bandwidth should purchase more data packs to restore access.
batch response: This repository is over its data quota. Account responsible for LFS bandwidth should purchase more data packs to restore access.
error: failed to fetch some objects from 'https://github.com/NVIDIA-Omniverse/IsaacGymEnvs.git/info/lfs'
```

I can reproduce the issue via:

```shell
git clone https://github.com/NVIDIA-Omniverse/IsaacGymEnvs.git
cd IsaacGymEnvs/
git lfs fetch
```

Is there any solution to this?

PS: in case you have some certificate issues, run `git config http.sslVerify false` before running `git lfs fetch`.

gavrielstate, 2021-11-05:
Thanks for letting us know. We're looking into this now.

gavrielstate, 2021-11-05:
Should be fixed now. We've removed the assets from LFS

closed 2021-11-05 by gavrielstate