isaac-sim/IsaacGymEnvs #6: errors when using IsaacGymEnvs as a single module

quark2019, 2021-11-11:
After installed IsaacGym (preview3) and IsaacGymEnvs, I can run each demo in IsaacGymEnvs OK.
But there are some errors when I trying to import some function / class in IsaacGymEnvs from my project.

--
Issue 1:
After running the command:
$ cd IsaacGymEnvs ; pip install .

IsaacGymEnvs's code doesnot copy into python/**/site-package/,  but only an egg file.

--
Issue 2:
By copying IsaacGymEnvs/isaacgymenvs to  python/**/site-package/ manually, it cannot be import by third project outside.
for example:
from isaacgymenvs.tasks.base.vec_task import  VecTask
raise an error:
.../site-packages/isaacgymenvs/tasks/__init__.py, line 30, in <module>
 from tasks.allegro_hand import AllegroHand
ModuleNotFoundError: No module named 'tasks'

Much thanks!

gavrielstate, 2021-11-11:
For (1), try:
`pip install -e .`

For (2), we haven't really been planning on this framework as something that would be used directly in a library-like fashion.  You are probably best to fork and create your own environments separately at the moment.

closed 2021-11-11 by gavrielstate

quark2019, 2021-11-12:
That's fine. Thanks again for your reply.