isaac-sim/IsaacGymEnvs #7: Multi-agent reinforcement learning algorithms

reso1, 2021-11-18:
I'm trying to use Isaac Gym for multi-agent reinforcement learning (MARL), is there any future plan on this? 

If not, can you give any suggestions on how should I integrate MARL algorithms into Issac Gym? Should I start from rl_games repo or maybe integrate other MARL-supported repos?

Thanks for your help!

Denys88, 2021-11-30:
rl_games supports MARL.
You can use independent PPO or with central value.
Starcraft multi agent is the good example: https://github.com/Denys88/rl_games/blob/master/rl_games/envs/smac_env.py
In env you just need to implement this function: 

```
    def get_number_of_agents(self):
        return self.n_agents 
```

If you want to use central_value you need to set 
`use_central_value = True `
and return dict:
```
{
'obs' : obs,
'state' : state
}
```

for observations.
In this case obs shape will be (envs_count * agents_count, *) and state shape is (envs_count, *).
You can take a look config example for this use case: 
https://github.com/Denys88/rl_games/blob/master/rl_games/configs/smac/3m.yaml

I think if you try to create simple MA env from ant, for example every leg can be agent :) I can help to adapt and test it with rl_games.
Thanks,
Denys

reso1, 2021-12-01:
Thanks for your reply and kind advices! I will try to implement this idea in IsaacGym and share further results.

I've read the rl_games MARL part code on SMAC env, it supports MARL indeed. But I have 2 further questions on the MARL algorithm implementation of rl_games, based on my code reading and understanding.

1. Is the multi-agent PPO implementation in rl_games identical to MAPPO algorithm in this [paper](http://bair.berkeley.edu/blog/2021/07/14/mappo/)?
2. Can the current rl_games MARL algorithm support heterogeneous agents w/ different action/observation space (i.e. different actor/critic nets)?

Denys88, 2021-12-01:
1. Almost yes. A little bit other value normalization code. And I didn't apply death masking. 

2. My goal was to achieve maximum performance on gpu so I didn't add support for cases like this directly.  But if you are using discrete action space you can provide different action masks to disable  some actions. And right now only different obs space for agent and critic is supported.

reso1, 2021-12-02:
Thanks a lot for explanations! I've implemented the MA ant env (each leg corresponds to an agent), however the training result is quite bad for ant, maybe using MAPPO for multi-joints robot control is that suitable? Anyway thanks again!

Denys88, 2021-12-02:
Could you try to use central value with a whole observation?
But I believe this case should work fine even with independent ppo. Can you share your env?

reso1, 2021-12-03:
Hi Denys, actually I used central value for the task however not getting good results, maybe there are some problems in my code and I'm very happy if you could help to check my code. 

I've forked IsaacGymEnvs and committed my MA_Ant env, you can check [this link](https://github.com/reso1/IsaacGymEnvs) and test it, the registered env task name is _MA_Ant_.

Apart from the new env class _ma_ant_, there are also 2 changes in the original IsaacGymEnvs repo:
1. The _ma_ant_ class is inherited from the _ma_vec_task_ class, where only function _allocate_buffers_ is modified to adapt to the buffer shape changes.
2. In the _get_env_info_ function in _RLGPUEnv_ class, number of agents and use_central_value are added to the info dict.

Denys88, 2021-12-03:
Thanks, Ill play with it tomorrow evening.
Just wondering what best score did you achieve?

reso1, 2021-12-03:
The best reward of _MA_Ant_ is below 10 (_Ant_ env is around 4k), but I did not fine-tune training parameters used in _MA_Ant_, so I don't whether there are some problems in my code, or it's just not suitable to consider each leg as an agent in MAPPO.

Denys88, 2021-12-04:
@reso1  I've found bug in your code:
expand_env_ids was buggy, so you actually always returned done=True for 3of4 legs.
I rewrote it in that way to make sure it works:
```
@torch.jit.script
def expand_env_ids(env_ids, n_agents):
    # type: (Tensor, int) -> Tensor
    device = env_ids.device
    agent_env_ids = torch.zeros((n_agents * len(env_ids)), device=device, dtype=torch.long)
    agent_env_ids[0::4] = env_ids * n_agents + 0
    agent_env_ids[1::4] = env_ids * n_agents + 1
    agent_env_ids[2::4] = env_ids * n_agents + 2
    agent_env_ids[3::4] = env_ids * n_agents + 3
    return agent_env_ids
```
And I added one more change to the compute_ant_observations:
```
    obs_idxs = torch.eye(4, dtype=torch.float32, device=self.device)
    obs_idxs = obs_idxs.repeat(state.size()[0], 1)
    obs = state.repeat_interleave(repeats, dim=0)
    obs = torch.cat([obs, obs_idxs], dim=-1)
```
So legs see a little bit different observations, I think it is not a must for your case but I thought it was the initial issue so didn't try 
And achieved 4k+ reward in less than 2 minutes without using central value(any way all legs see the same observations).
![image](https://user-images.githubusercontent.com/1936835/144692600-1c71be63-5b3c-4d13-82b4-f800ebabcdbb.png
As you see movement is not as good and we need some tuning to make it perfect but it works somehow.
[image]
)

reso1, 2021-12-05:
Cool, thanks for your kind help! 

BTW, if you have any plan on formally integrating MARL on IsaacGym using rl_games, I'm very glad to help it :)

closed 2021-12-05 by reso1

bycn, 2021-12-18:
Hi @reso1, could you please give me some pointers if I wanted to fork your MA ant repo to add a simple mat to the environment? In particular I would like to recreate the RoboSumo environment https://github.com/openai/robosumo . Thank you!

reso1, 2021-12-22:
Hi Bryan, the following is the pointer to the MA ant repo, you can check
the closed issue in IsaacGymEnvs where I mentioned details about this
environment.

https://github.com/reso1/IsaacGymEnvs

Enjoy!

Frank-Dz, 2022-06-15:
> Hi Bryan, the following is the pointer to the MA ant repo, you can check the closed issue in IsaacGymEnvs where I mentioned details about this environment. https://github.com/reso1/IsaacGymEnvs Enjoy! Bryan Chen ***@***.***> 于2021年12月18日周六 10:41写道：
> […](#)

Hi ~ Are you going to share the repo? I am excited to play with it:D