isaac-sim/IsaacGymEnvs #8: Do we able to use image as input for IsaacGym ?

sizhe-wang, 2021-12-03:
Dear, author:
    we have seen  resnet_actor_critic setting for training the RL algorithm.   Does IsaacGym support image type input ? Thanks advance. looking forward your reply !

ViktorM, 2021-12-10:
Hi @sizhe-wang,

Yes, Isaac Gym supports camera sensors. We don't currently have examples of RL training environments using camera input but might consider adding a simple one as it is a pretty common request to provide an example. @gavrielstate for visibility.

gavrielstate, 2022-01-02:
We're hoping to add an additional basic example using camera input for RL control later this winter.

noshaba, 2022-02-02:
@gavrielstate How about controller inputs? :)

gavrielstate, 2022-02-03:
We won't be adding direct support for controllers to Isaac Gym I'm afraid, but it should be available now within Isaac Sim. We will have more RL capabilities in future versions of Isaac Sim that include the Gym-style tensor physics APIs.

erwincoumans, 2022-09-03:
See my fork: https://github.com/erwincoumans/IsaacGymEnvs/tree/cartpole_camera
It has task=CartpoleCamera (thanks to @ViktorM )

closed 2022-09-03 by erwincoumans