isaac-sim/IsaacGymEnvs #9: Pointers to create a MA environment

bycn, 2021-12-18:
Hi all, 
I'm trying to recreate the RoboSumo environment https://github.com/openai/robosumo . Could someone give me an overview / code pointers to create this? It would essentially be two Ants whos actions affect each other, along with a simple mat texture on the bottom to indicate the "ring".

ViktorM, 2021-12-18:
Hi @bycn,

Currently, we don't have an example of multi-agent and self-play environments in Isaac Gym. But I've already had the re-implementation of robosumo working in Isaac Gym and training with rl_games internally. It could be trained in less than 2 hours on a single GPU (PC). We plan to release it in one of the next updates to IsaacGymEnvs.

bycn, 2021-12-18:
@ViktorM Great, thanks, looking forward to the release :)

closed 2021-12-18 by bycn