isaac-sim/IsaacGymEnvs #13: how can i train it on remote system.

zituka, 2022-01-14:
Hi, thx for the great job on  IsaacGymEnvs!
I don’t own a local desktop machine, I rely on remote cluster for GPUs. I have headless docker + ssh./ conda + ssh. ,(i tried VNC, but it seems not good on isaac gym)
All the python/examples show segfault, RL env report the same segfault core dump error. I looked up a lot of methods but nothing worked. How can i train it just on terminal?

HaozhiQi, 2022-01-14:
Have you tried to set `headless: True` in `cfg/config.yaml`?

zituka, 2022-01-15:
> 

many thanks for your reply, i forgot that.. it looks good now!

closed 2022-01-15 by zituka

abayomi185, 2022-07-06:
Hi, is it possible to view the graphics environment on the remote system using VNC, RDP or perhaps from Docker?