isaac-sim/IsaacGymEnvs #16: Missing / Changed Environments

noshaba, 2022-02-02:
Hi,

It seems like several environments mentioned in the [white papers](https://arxiv.org/pdf/2108.10470.pdf) are not included, e.g. Humanoid Character Animation (missing), Franka Cube Stacking (task changed to cabinet opening for Franka), also the terrain for the ANYmal Terrain version in the white papers is more complex than in the published code but I guess that can be changed easily. I am mostly bummed out about not including the Humanoid Character Animation environment with AMP. Any reason why these were not included/changed?

gavrielstate, 2022-02-03:
We are putting the final touches on AMP and should have a version quite soon. There's been a fair bit of work to clean it up - for example, right now we're dealing with writing scripts to do motion capture conversions since the original research code we were working with was using one-off hacky conversions that were hard to duplicate.

The OSC examples should come after that.

The Anymal terrain example we include in IsaacGymEnvs is similar to, but not identical to the one used in the Learning to Walk in Minutes paper. You can find the code used for that paper in this repository: https://github.com/leggedrobotics/legged_gym

noshaba, 2022-02-03:
Interesting, thank you :) 
Which motion capture formats will you be supporting?

janisa9, 2022-02-14:
@gavrielstate what about Franka Cube Stacking, will it also be available?

gavrielstate, 2022-03-22:
The poselib library we ship as part of AMP will have conversion scripts for FBX animations.

Franka Cube Stacking will come after the AMP release - we realized that there was a URDF model change in the standalone work we'd done outside IsaacGymEnvs

lorenmt, 2022-04-09:
Any updates for the Frank's tasks? Love to hear any updates if possible!