isaac-sim/IsaacGymEnvs #17: Documentation needs to be updated to Isaac Preview 3

noshaba, 2022-02-02:
Hi,

The documentation, e.g. in https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/blob/main/docs/framework.md is not up-to-date anymore with the new version.

gavrielstate, 2022-02-03:
Can you be more specific about what you think needs to be updated?

noshaba, 2022-02-03:
For example it states that the __init__ method only takes 3 arguments - the config file, sim_device and headless. Yet, in the examples, they also include graphics_device_id. Also the parameters required for the config files and what they mean and which options you have for them would be nice to have written somewhere in a more formal and organized way.

noshaba, 2022-02-03:
So at least a minimum hello world example, similar to the Hello World / Robot examples in Isaac Sim.

noshaba, 2022-02-07:
Another improvement for the docs could be to add that jit functions should all have unique names. It doesn't work if the same function name for the reward and observation function are used in different tasks but the error messages you get don't indicate that. Only says that the task X could not be imported.

closed 2022-04-10 by noshaba