isaac-sim/IsaacGymEnvs #20: AssertionError when changin number of environments

AntonBock, 2022-02-16:
Hello, I have suddenly started getting and AssertionError, when changing the numEnvs in the RL examples. I thought my computer was messed up, so I reinstalled Ubuntu 20.04. My Isaac Gym is therefore running on a fresh system, however the error persists. 

**When I change the numEnvs parameter in the Cartpole.yaml from 512 to 256, i get the following error:**

```
Error executing job with overrides: ['task=Cartpole']
Traceback (most recent call last):
  File "train.py", line 112, in launch_rlg_hydra
    'play': cfg.test,
  File "/home/anton/anaconda3/envs/rlgpu/lib/python3.7/site-packages/rl_games/torch_runner.py", line 139, in run
    self.run_train()
  File "/home/anton/anaconda3/envs/rlgpu/lib/python3.7/site-packages/rl_games/torch_runner.py", line 122, in run_train
    agent = self.algo_factory.create(self.algo_name, base_name='run', config=self.config)
  File "/home/anton/anaconda3/envs/rlgpu/lib/python3.7/site-packages/rl_games/common/object_factory.py", line 15, in create
    return builder(**kwargs)
  File "/home/anton/anaconda3/envs/rlgpu/lib/python3.7/site-packages/rl_games/torch_runner.py", line 23, in <lambda>
    self.algo_factory.register_builder('a2c_continuous', lambda **kwargs : a2c_continuous.A2CAgent(**kwargs))
  File "/home/anton/anaconda3/envs/rlgpu/lib/python3.7/site-packages/rl_games/algos_torch/a2c_continuous.py", line 18, in __init__
    a2c_common.ContinuousA2CBase.__init__(self, base_name, config)
  File "/home/anton/anaconda3/envs/rlgpu/lib/python3.7/site-packages/rl_games/common/a2c_common.py", line 973, in __init__
    A2CBase.__init__(self, base_name, config)
  File "/home/anton/anaconda3/envs/rlgpu/lib/python3.7/site-packages/rl_games/common/a2c_common.py", line 171, in __init__
    assert(self.batch_size % self.minibatch_size == 0)
AssertionError
``` 

Any help is much appreciated

AntonBock, 2022-02-17:
For some reason, when running the examples using [this wrapper](https://github.com/Toni-SM/skrl), I can change the number of environments without problem.

AntonBock, 2022-02-18:
I think i found a/the solution in the .yaml files under IsaacGymEnvs/isaacgymenvs/cfg/train. The minibatch_size must be dividable with num_envs multiplied with the horizon_length. This at least fixes it for me.

gavrielstate, 2022-03-22:
Yes, this is an rl_games requirement, so SKRL would indeed be different.

We will likely update to a newer version of rl_games where this may change at some point later this year.

closed 2022-03-22 by gavrielstate

LiHaochuan123, 2024-10-16:
I found that num_envs must be the several times default numEnvs in the .yaml files under IsaacGymEnvs/isaacgymenvs/cfg/task. I found this issue when I attempted to run 
```bash
python train.py task=Cartpole num_envs=512
```