isaac-sim/IsaacGymEnvs #27: Virtual environment installations can't run joint_monkey.py

tsbertalan, 2022-04-06:
On a machine without GPU:
```bash
lsb_release -a  # Ubuntu 20.04.4 LTS
python3 --version # Python 3.8.10
python3 -m venv ~/.virtualenvs/isaacgym
source ~/.virtualenvs/isaacgym/bin/activate
cd python
pip install -e .
cd examples
python joint_monkey.py
```
This gives:
```
*** Warning: failed to preload CUDA lib
*** Warning: failed to preload PhysX libs
Importing module 'gym_38' (/home/tsbertalan/Dropbox/Projects/ODEs for RL/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_38.so)
[Error] [carb] [Plugin: libcarb.gym.plugin.so] Could not load the dynamic library from /home/tsbertalan/Dropbox/Projects/ODEs for RL/isaacgym/python/isaacgym/_bindings/linux-x86_64/libcarb.gym.plugin.so. Error: libcuda.so.1: cannot open shared object file: No such file or directory
[Warning] [carb] Potential plugin preload failed: /home/tsbertalan/Dropbox/Projects/ODEs for RL/isaacgym/python/isaacgym/_bindings/linux-x86_64/libcarb.gym.plugin.so
Setting GYM_USD_PLUG_INFO_PATH to /home/tsbertalan/Dropbox/Projects/ODEs for RL/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
[Error] [carb] Failed to acquire interface: [carb::gym::Gym v0.1], by client: carb.gym.python.gym_38 (plugin name: (null))
Traceback (most recent call last):
  File "joint_monkey.py", line 59, in <module>
    gym = gymapi.acquire_gym()
RuntimeError: Failed to acquire interface: carb::gym::Gym (pluginName: nullptr)
```

I then go to a Ubuntu 18.04.5 LTS machine with CUDA 11.1 and an RTX 2080 Ti with the 455.45.01 driver (per `nvidia-smi`).
```bash
sudo apt install python3.8 python3.8-dev  # from http://ppa.launchpad.net/deadsnakes/ppa/ubuntu bionic/main amd64 Packages
mkvirtualenv --python=/usr/bin/python3.8 isaacgym  # 3.8.13
workon isaacgym
cd python
pip install -e .
cd examples
python joint_monkey.py
```
and get
```
Importing module 'gym_38' (/home/tsbertalan/Dropbox/Projects/ODEs for RL/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_38.so)
Setting GYM_USD_PLUG_INFO_PATH to /home/tsbertalan/Dropbox/Projects/ODEs for RL/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
WARNING: Forcing CPU pipeline.
Not connected to PVD
/buildAgent/work/45f70df4210b2e3e/source/cudamanager/src/CudaContextManager.cpp (635) : warning : CUDA driver version is 11010, expected at least 11040.

/buildAgent/work/45f70df4210b2e3e/source/cudamanager/src/CudaContextManager.cpp (370) : info : NVIDIA Release 411 graphics driver and above is required for GPU acceleration.

[Warning] [carb.gym.plugin] Failed to create a valid PhysX CUDA Context Manager. Falling back to CPU.
Physics Engine: PhysX
Physics Device: cpu
GPU Pipeline: disabled
Segmentation fault (core dumped)
```

I also try with Miniconda on the 18.04 machine:

```bash
conda activate base
./create_conda_env_rlgpu.sh
conda activate rlgpu
cd python/examples/
python joint_monkey.py
```

I get
```
Importing module 'gym_37' (/home/tsbertalan/Dropbox/Projects/ODEs for RL/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_37.so)
Traceback (most recent call last):
  File "joint_monkey.py", line 19, in <module>
    from isaacgym import gymapi, gymutil
  File "/home/tsbertalan/Dropbox/Projects/ODEs for RL/isaacgym/python/isaacgym/__init__.py", line 5, in <module>
    from isaacgym import gymapi
  File "/home/tsbertalan/Dropbox/Projects/ODEs for RL/isaacgym/python/isaacgym/gymapi.py", line 104, in <module>
    _import_active_version()
  File "/home/tsbertalan/Dropbox/Projects/ODEs for RL/isaacgym/python/isaacgym/gymapi.py", line 63, in _import_active_version
    module = importlib.import_module(package_path)
  File "/home/tsbertalan/miniconda3/envs/rlgpu/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: libpython3.7m.so.1.0: cannot open shared object file: No such file or directory
```

I am able to run the Omniverse installer (on the first machine), do its 2 or 3GB GUI installer for "Isaac SIm", and then launch that GUI. But this seems like a totally different thing from the gym environments here, and the terminal it shows displays [a lot of errors](https://gist.github.com/tsbertalan/12012f3a80c1133c921cc0b61d001782), though a GUI appears and is clickable.

lapp0, 2022-04-11:
Another thread tracking this issue https://forums.developer.nvidia.com/t/wsl2-and-isaac-gym-problem/192069/10

kiwi-sherbet, 2022-04-24:
I am using pyenv-virtualenv and having the same issues. Have you solved the issue?

tsbertalan, 2022-04-29:
I haven't--I've moved to looking at other simulation platforms for my needs for now, and will revisit this if there's action from the authors towards better packaging.

kiwi-sherbet, 2022-04-29:
I figured out how to solve it. 
The following installation command resolved the issue. 

```
ISAAC_VENV_NAME='Isaac_gym_env'
ISAACGYM_PATH='./isaacgym'
GYM_EXAMP_PATH='./IsaacGymEnvs'

CONFIGURE_OPTS=--enable-shared pyenv install 3.7.10
pyenv virtualenv 3.7.10 $ISAAC_VENV_NAME
pyenv activate $ISAAC_VENV_NAME

pip3 install -r ./requirements.txt # Install dependencies. 
pip3 install -e "$ISAACGYM_PATH/python"
pip3 install -e "$GYM_EXAMP_PATH"
```

tsbertalan, 2022-05-03:
@kiwi-sherbet Do you have a CUDA GPU in that machine? It's not clear to me whether this thing is supposed to be able to run CPU-only or not.

kiwi-sherbet, 2022-05-03:
Yes, I am using CUDA 11.2. I have not tried running it with CPU only.

tsbertalan, 2022-05-04:
I think that does it, though I won't really know until later; see below. The directory `$PROJECT_DIR` referenced below contains isaacgym and IsaacGymEnvs.

I ran the following on our lab's Ubuntu 18.04 server with an RTX 2080ti and driver 455.45.01 with CUDA 11.1 per `nvidia-smi`:

```bash
git clone https://github.com/pyenv/pyenv.git ~/.pyenv
cd ~/.pyenv && src/configure && make -C src
echo 'export PYENV_ROOT="$HOME/.pyenv"' >> ~/.bashrc
echo 'command -v pyenv >/dev/null || export PATH="$PYENV_ROOT/bin:$PATH"' >> ~/.bashrc
echo 'eval "$(pyenv init -)"' >> ~/.bashrc
echo 'export PYENV_ROOT="$HOME/.pyenv"' >> ~/.profile
echo 'command -v pyenv >/dev/null || export PATH="$PYENV_ROOT/bin:$PATH"' >> ~/.profile
echo 'eval "$(pyenv init -)"' >> ~/.profile
git clone https://github.com/pyenv/pyenv-virtualenv.git $(pyenv root)/plugins/pyenv-virtualenv
cd "$PROJECT_DIR"
ISAAC_VENV_NAME='Isaac_gym_env'
ISAACGYM_PATH='./isaacgym'
GYM_EXAMP_PATH='./IsaacGymEnvs'
sudo apt-get install libffi-dev  # Actually, first I did the next 3 commands without this, and the first pip install failed on pycollada==0.6. So, I had to
#   pyenv deactivate
#   pyenv uninstall 3.7.10/envs/Isaac_gym_env 
#   pyenv uninstall 3.7.10
# and then redo the pyenv install 3.7.10
CONFIGURE_OPTS=--enable-shared pyenv install 3.7.10
pyenv virtualenv 3.7.10 $ISAAC_VENV_NAME
pyenv activate $ISAAC_VENV_NAME
```

Activating the env here failed; it instructed me to restart the terminal.

```bash
pip3 install -r ./requirements.txt # This is my own file, but whatever.
pip3 install -e "$ISAACGYM_PATH/python"
```

Finally, I did the test script, as `python isaacgym/python/examples/joint_monkey.py`. This time, it failed due to OOM on the GPU, which is to be expected because we have a visitor using all the GPU RAM as he usually does. I'll ask him nicely to get off and then try again later. But it's probably fixed now?

3DWen, 2024-10-12:
RuntimeError: Failed to acquire interface: carb::gym::Gym (pluginName: nullptr)
这个问题，请问您解决了吗？

Yixun-Hu, 2025-08-04:
When I tried to run 1080 balls simulation, I get this error: [1]    948387 segmentation fault (core dumped)  python 1080_balls_of_solitude.py