isaac-sim/IsaacGymEnvs #35: RuntimeError: Failed to acquire interface: carb::gym::Gym (pluginName: nullptr)

emigmo, 2022-04-18:
Hi, isaacgym can not run in  A100-SXM-80GB ?
When I run the test code in this type of GPUs, I get the following note:

> Exact experiment name requested from command line: Ant
/home/yangchao/anaconda3/envs/rl/lib/python3.7/site-packages/gym/spaces/box.py:84: UserWarning: WARN: Box bound precision lowered by casting to float32
  logger.warn(f"Box bound precision lowered by casting to {self.dtype}")
[Error] [carb] Failed to acquire interface: [carb::gym::Gym v0.1], by client: carb.gym.python.gym_37 (plugin name: (null))
Error executing job with overrides: []
Traceback (most recent call last):
  File "train.py", line 127, in launch_rlg_hydra
    'play': cfg.test,
  File "/home/yangchao/anaconda3/envs/rl/lib/python3.7/site-packages/rl_games/torch_runner.py", line 139, in run
    self.run_train()
  File "/home/yangchao/anaconda3/envs/rl/lib/python3.7/site-packages/rl_games/torch_runner.py", line 122, in run_train
    agent = self.algo_factory.create(self.algo_name, base_name='run', config=self.config)
  File "/home/yangchao/anaconda3/envs/rl/lib/python3.7/site-packages/rl_games/common/object_factory.py", line 15, in create
    return builder(**kwargs)
  File "/home/yangchao/anaconda3/envs/rl/lib/python3.7/site-packages/rl_games/torch_runner.py", line 23, in <lambda>
    self.algo_factory.register_builder('a2c_continuous', lambda **kwargs : a2c_continuous.A2CAgent(**kwargs))
  File "/home/yangchao/anaconda3/envs/rl/lib/python3.7/site-packages/rl_games/algos_torch/a2c_continuous.py", line 18, in __init__
    a2c_common.ContinuousA2CBase.__init__(self, base_name, config)
  File "/home/yangchao/anaconda3/envs/rl/lib/python3.7/site-packages/rl_games/common/a2c_common.py", line 980, in __init__
    A2CBase.__init__(self, base_name, config)
  File "/home/yangchao/anaconda3/envs/rl/lib/python3.7/site-packages/rl_games/common/a2c_common.py", line 85, in __init__
    self.vec_env = vecenv.create_vec_env(self.env_name, self.num_actors, **self.env_config)
  File "/home/yangchao/anaconda3/envs/rl/lib/python3.7/site-packages/rl_games/common/vecenv.py", line 282, in create_vec_env
    return vecenv_config[vec_env_name](config_name, num_actors, **kwargs)
  File "train.py", line 95, in <lambda>
    lambda config_name, num_actors, **kwargs: RLGPUEnv(config_name, num_actors, **kwargs))
  File "/home/yangchao/Workspace/1-isaacgym/IsaacGymEnvs/isaacgymenvs/utils/rlgames_utils.py", line 159, in __init__
    self.env = env_configurations.configurations[config_name]['env_creator'](**kwargs)
  File "train.py", line 98, in <lambda>
    'env_creator': lambda **kwargs: create_rlgpu_env(**kwargs),
  File "/home/yangchao/Workspace/1-isaacgym/IsaacGymEnvs/isaacgymenvs/utils/rlgames_utils.py", line 91, in create_rlgpu_env
    headless=headless
  File "/home/yangchao/Workspace/1-isaacgym/IsaacGymEnvs/isaacgymenvs/tasks/ant.py", line 70, in __init__
    super().__init__(config=self.cfg, sim_device=sim_device, graphics_device_id=graphics_device_id, headless=headless)
  File "/home/yangchao/Workspace/1-isaacgym/IsaacGymEnvs/isaacgymenvs/tasks/base/vec_task.py", line 181, in __init__
    self.gym = gymapi.acquire_gym()
RuntimeError: Failed to acquire interface: carb::gym::Gym (pluginName: nullptr)
Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.

zalo, 2022-05-03:
I ran into this issue too... I suspect that downloading Omniverse Kit would have updated Carbonite enough to load it, but Omniverse is having a fit right now there too:
https://forums.developer.nvidia.com/t/installation-error-cannot-sign-url-for-download-http403/213238

I eventually gave up on that and found this path:
https://forums.developer.nvidia.com/t/failed-to-acquire-interface/178379/14

Running the whole thing from docker seems to be the safest way to go about it; I even managed to get a HumanoidAMP model converging (just need to run `run.sh` with a 0 afterward to  get it in display mode, and omit the `headless` param on `train.py`).

heartInsert, 2024-07-08:
Did you solve it ? I  meet the same problem.