isaac-sim/IsaacGymEnvs #39: The AMP method in IsaacGym is not the same as the AMP method in the paper

Timkeeper2018, 2022-04-25:
I see that the GAN loss function used in your AMP algorithm example is not the same as the loss function mentioned in the AMP paper. You are using something like        "-E[log(D(s,s'))]-E[log(1-D(s,s'))]"          as the formula (5) mentioned in the AMP paper, But they used formula (6) from LSGAN.
  The reward function and gradient penalty function are also different from those mentioned in the AMP paper.
  I'm wondering why you guys have a different approach than AMP? Or am I looking at something wrong?

mJones00, 2022-05-22:
Maybe they found out that it work better with the original GAN loss since they also change it in their new ASE paper

amy12xx, 2022-09-14:
It looks like they train both discriminator and policy [together](https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/blob/ca7a4fb762f9581e39cc2aab644f18a83d6ab0ba/isaacgymenvs/learning/amp_continuous.py#L323), at the same rate (single optimizer and combined loss). Both in AMP and ASE. Any idea why that is done? In both papers' pseudocode, they seem to suggest this is done separately.