isaac-sim/IsaacGymEnvs #41: Creating a T-Pose .npy file with Poselib

noshaba, 2022-05-05:
Hi,

I’m failing to create a T-Pose .npy. I want to retarget the SFU Wushu motion to the AMP skeleton. As I do not have any MJCF file from them but only FBX, I tried to extract the T-Pose from the FBX and saved it as another FBX. However, when I run fbx_importer on it it creates a SkeletonMotion but the retargeting expects a SkeletonState for the T-Pose. In the example a SkeletonState is only created from a MJCF file however.

Best,
Noshaba

mJones00, 2022-05-22:
In my case, I have a t-pose at the beginning of the motion in the fbx file, so I found a crop method for SkeletonMotion and changed it so that it return a SkeletonState instead, the cropped the beginning of my motion to be my t-pose.
so you can change the crop method for the SkeletonMotion class in poselib/skeleton/skeleton3d.py with something like:

`
def crop(self, start: int, end: int, fps: Optional[int] = None, return_skel_state: bool = False):
        """
        Crop the motion along its last axis. This is equivalent to performing a slicing on the
        object with [..., start: end: skip_every] where skip_every = old_fps / fps. Note that the
        new fps provided must be a factor of the original fps. 

        :param start: the beginning frame index
        :type start: int
        :param end: the ending frame index
        :type end: int
        :param fps: number of frames per second in the output (if not given the original fps will be used)
        :type fps: int, optional
        :rtype: SkeletonMotion
        """
        if fps is None:
            new_fps = int(self.fps)
            old_fps = int(self.fps)
        else:
            new_fps = int(fps)
            old_fps = int(self.fps)
            assert old_fps % fps == 0, (
                "the resampling doesn't support fps with non-integer division "
                "from the original fps: {} => {}".format(old_fps, fps)
            )
        skip_every = old_fps // new_fps
        if return_skel_state:
            return SkeletonState.from_rotation_and_root_translation(
            skeleton_tree=self.skeleton_tree,
            t=self.root_translation[start:end:skip_every],
            r=self.local_rotation[start:end:skip_every],
            is_local=True
          )
        else:
            return SkeletonMotion.from_skeleton_state(
            SkeletonState.from_rotation_and_root_translation(
                skeleton_tree=self.skeleton_tree,
                t=self.root_translation[start:end:skip_every],
                r=self.local_rotation[start:end:skip_every],
                is_local=True
            ),
            fps=self.fps
            )
`
Then use it in fbx_importer.py with:
`
motion.crop(0, 1, return_skel_state=True).to_file("data/FBX_tpose.npy")
`

Frank-Dz, 2022-06-10:
> Hi,
>
> I’m failing to create a T-Pose .npy. I want to retarget the SFU Wushu motion to the AMP skeleton. As I do not have any MJCF file from them but only FBX, I tried to extract the T-Pose from the [...]

Hi, Do you have a T-pose npy? Could you share it with me:D Thanks!