isaac-sim/IsaacGymEnvs #42: Getting image observations from environment

shivakanthsujit, 2022-05-08:
Hello,

If I am running on a headless server, is there anyway for me to get image observations from the environment so that I can generate/save a video of the agent's performance? Something like gym's `env.render("rgb_array")` that I can use when I have set `headless=True`.

Thanks!

42jaylonw, 2022-05-23:
Hi,

What I did was create a camera sensor:
`self.local_pos = gymapi.Vec3([1., -1, 1.])`
`self.target_pos = gymapi.Vec3([0., 0., 0.])`
`camera_handle = self.gym.create_camera_sensor(env_handle, self.cfg.camera_props)`
`self.gym.set_camera_location(camera_handle, env_handle, self.local_pos, self.target_pos)`

and load image whenever is needed:
refresh state first:
`self.gym.render_all_camera_sensors(self.sim)`
`self.gym.fetch_results(self.sim, True)`
`self.gym.step_graphics(self.sim)`
get image tensor:
`img = self.gym.get_camera_image_gpu_tensor(self.sim, env_handle, cam_handle, gymapi.IMAGE_COLOR)`
`img_tensor = gymtorch.wrap_tensor(img)`

In this way, you can get image whenever headless is `True` or `False`

you can find more instructions in your isaacgym docs: `$your_isaacgym/isaacgym/docs/index.html`

42jaylonw, 2022-11-23:
In my point of view,
isaacgym does not support adding a camera sensor after the initialization or `self.gym.prepare_sim(self.sim)`.
In other words,  if you are using the built-in making env making function `isaacgymenvs.make` you cannot add a camera sensor. You should either create your camera sensor earlier.

Or you can check example 2 from my wrapper, it's more easier to add a camera sensor and load image data.
https://github.com/42jaylonw/shifu/

Rowing0914, 2023-09-08:
hmm, this happens even before `gym.prepare_sim`. do you know why?

mxllc, 2023-11-14:
It's things of the initialization of func **gym.create_sim**.
You need to specify the parameter **graphics_device**.
If this parameter is _-1_, then you would get following results.
> SetCameraLocation: Error: could not find camera with handle -1 in environment 0

It's the parameter for GPU id for rendering.
Here is a brief introduction of this function.
[image]