isaac-sim/IsaacGymEnvs #46: Requiring simpler API for designating physical parameters

major-second, 2022-05-21:
Dear Developers in NVIDIA:
   Thank you for your impressive work on this repository!
   I'm launching a transfer learning research program in robotics. In the program, I should train policies in environments with different physical parameters and then test them in a certain one with specified parameters. However, I found it not so straightforward to designate the parameter vector using the current IsaacGymEnvs codebase --- I could only control the range of randomization of each component. Further, some parameters (e.g., the friction coefficients of all the 7 DOFs) share the same randomization range, so never could I specify an interval of length 0 to emulate the specification process.
   I would appreciate it if you could update the repository with some finer-grained domain randomization APIs where I could specify the parameters directly!

major-second, 2022-05-21:
We would appreciate it if this function
```
(https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/tree/main/isaacgymenvs/tasks/base)/vec_task.py 

Line 512
    def apply_randomizations(self, dr_params):
```
could have a counterpart: apply_parameters(self, params)

quantumiracle, 2022-05-21:
Hi, 

I also found it difficult to apply (domain randomization) DR part in some research projects. Specifically, (1) when I want to pass in certain parameter values for each variable for one or several episodes, or (2) get the values out from the internal DR process for a certain episode. 

One thing I notice is that Isaacgym envs apply a different reset procedure, which is asynchronous for all parallel evn, so I'm not sure if it is possible to achieve the functions I described above.

Thanks