isaac-sim/IsaacGymEnvs #52: Segmentation fault (core dumped)

nuomizai, 2022-06-27:
When I ran the example file `python train.py task=Ant headless=True`, I got an error as `Segmentation fault (core dumped)`. The full output is as follows:

```
Importing module 'gym_38' (/home/liuchi/zhaoyinuo/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_38.so)
Setting GYM_USD_PLUG_INFO_PATH to /home/liuchi/zhaoyinuo/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
PyTorch version 1.11.0+cu102
Device count 8
/home/liuchi/zhaoyinuo/isaacgym/python/isaacgym/_bindings/src/gymtorch
Using /home/liuchi/.cache/torch_extensions/py38_cu102 as PyTorch extensions root...
Emitting ninja build file /home/liuchi/.cache/torch_extensions/py38_cu102/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
Loading extension module gymtorch...
/home/liuchi/zhaoyinuo/isaacgym/python/isaacgym/torch_utils.py:135: DeprecationWarning: `np.float` is a deprecated alias for the builtin `float`. To silence this warning, use `float` by itself. Doing this
 will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.float64` here.
Deprecated in NumPy 1.20; for more details and guidance: https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
  def get_axis_params(value, axis_idx, x_value=0., dtype=np.float, n_dims=3):
2022-06-27 10:44:06,084 - INFO - logger - logger initialized
<unknown>:3: DeprecationWarning: invalid escape sequence \*
Error: FBX library failed to load - importing FBX data will not succeed. Message: No module named 'fbx'
FBX tools must be installed from https://help.autodesk.com/view/FBX/2020/ENU/?guid=FBX_Developer_Help_scripting_with_python_fbx_installing_python_fbx_html
/home/liuchi/anaconda3/envs/hand/lib/python3.8/site-packages/torch/utils/tensorboard/__init__.py:4: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  if not hasattr(tensorboard, '__version__') or LooseVersion(tensorboard.__version__) < LooseVersion('1.15'):
train.py:66: UserWarning:
The version_base parameter is not specified.
Please specify a compatability version level, or None.
Will assume defaults for version 1.1
  @hydra.main(config_name="config", config_path="./cfg")
/home/liuchi/anaconda3/envs/hand/lib/python3.8/site-packages/hydra/_internal/defaults_list.py:251: UserWarning: In 'config': Defaults list is missing `_self_`. See https://hydra.cc/docs/upgrades/1.0_to_1.
1/default_composition_order for more information
  warnings.warn(msg, UserWarning)
/home/liuchi/anaconda3/envs/hand/lib/python3.8/site-packages/hydra/_internal/defaults_list.py:415: UserWarning: In config: Invalid overriding of hydra/job_logging:
Default list overrides requires 'override' keyword.
See https://hydra.cc/docs/next/upgrades/1.0_to_1.1/defaults_list_override for more information.

  deprecation_warning(msg)
/home/liuchi/anaconda3/envs/hand/lib/python3.8/site-packages/hydra/_internal/hydra.py:119: UserWarning: Future Hydra versions will no longer change working directory at job runtime by default.
See https://hydra.cc/docs/next/upgrades/1.1_to_1.2/changes_to_job_working_dir/ for more information.
  ret = run_job(
task:
    name: Ant
    physics_engine: physx
    env:
        numEnvs: 4096
        envSpacing: 5
        episodeLength: 1000
        enableDebugVis: False
        clipActions: 1.0
        powerScale: 1.0
        controlFrequencyInv: 1
        headingWeight: 0.5
        upWeight: 0.1
        actionsCost: 0.005
        energyCost: 0.05
        dofVelocityScale: 0.2
        contactForceScale: 0.1
        jointsAtLimitCost: 0.1
        deathCost: -2.0
        terminationHeight: 0.31
        plane:
            staticFriction: 1.0
            dynamicFriction: 1.0
            restitution: 0.0
        asset:
            assetFileName: mjcf/nv_ant.xml
        enableCameraSensors: False
    sim:
        dt: 0.0166
        substeps: 2
[... 176 lines cut ...]
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
|   3  NVIDIA GeForce ...  Off  | 00000000:3E:00.0 Off |                  N/A |
| 37%   26C    P8    25W / 350W |    272MiB / 24268MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
|   4  NVIDIA TITAN RTX    Off  | 00000000:88:00.0 Off |                  N/A |
| 41%   28C    P8     6W / 280W |    180MiB / 24220MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
|   5  NVIDIA TITAN RTX    Off  | 00000000:89:00.0 Off |                  N/A |
| 41%   28C    P8     8W / 280W |    180MiB / 24220MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
|   6  NVIDIA TITAN RTX    Off  | 00000000:B1:00.0 Off |                  N/A |
| 41%   30C    P8    13W / 280W |    180MiB / 24220MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
|   7  NVIDIA TITAN RTX    Off  | 00000000:B2:00.0 Off |                  N/A |
| 41%   28C    P8    19W / 280W |    180MiB / 24220MiB |      0%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
```

My platform is ubuntu18.04.

nicholas0717, 2022-07-01:
I met the same situations Segmentation fault, and all the words displayed in the window of Isaac Gym were too vague to distinguish. Have you solved it？

nuomizai, 2022-07-06:
> I met the same situations Segmentation fault, and all the words displayed in the window of Isaac Gym were too vague to distinguish. Have you solved it？

Sorry, @nicholas0717 . I haven't solved it. But I found the reason -- my graphic desktop was broken. Therefore, If you run issacgym on a headless ubuntu server, I recommand you to install ubuntu desktop by

```
sudo apt-get update
sudo apt-get install ubuntu-desktop
reboot
```

Murrol, 2022-10-18:
Thanks, @nuomizai. On ubuntu18.04 with Nvidia driver version 470, I solved the issue with this.
`sudo apt-get update  
sudo apt-get install --reinstall ubuntu-desktop  
sudo apt-get install unity  
sudo reboot`

HaozhiQi, 2022-10-19:
I also found `Segmentation fault (core dumped)` happen if I launch isaacgym in tmux window (it works well without tmux). Just FYI.

nuomizai, 2022-11-02:
I solved the `Segmentation fault` problem by changing the nvidia-driver to 510.85.02 and cuda version to 11.6.

closed 2022-11-02 by nuomizai

ameliacode, 2022-11-24:
For anyone tries to install ubuntu desktop due to this issue, make sure there is no warning window.
Just simply reinstall driver.
My computer had booting loop error+ethernet not working + driver crashed after reboot..

amirhnourian, 2023-06-05:
I am currently experiencing this issue! in Isaac sim example https://docs.omniverse.nvidia.com/app_isaacsim/app_isaacsim/tutorial_intro_workflows.html#
./python.sh standalone_examples/api/omni.isaac.franka/follow_target_with_rmpflow.py
I am on ubuntu 22.04 should I install Cuda drivers as well? I just installed 5.25.x.x

LJoson, 2023-10-23:
I have the same issue. After debugging, I found that the issue was actually due to the lack of Vulkan installation. The official Vulkan is more complete on Ubuntu 20.04+, so if there are no special requirements, you can consider using Ubuntu 20.04+. Otherwise, you may need to install Vulkan from source code.

kc-ustc, 2023-11-07:
I also met this problem when I run the example(p_**ython 1080_balls_of_solitude.py**_) with emote sever.But I can run the example in my personal computer.The  diffrence environment of are the version of cuda and driver.Why?So confused.

kc-ustc, 2023-11-07:
I also met this problem when I run the example(**_p_ython 1080_balls_of_solitude.py__**) with emote sever.But I can run the example in my personal computer.The diffrence of environment  are the version of cuda and nvidia driver.Why?So confused.

kc-ustc, 2023-11-07:
I also met this problem when I run the example(p_ython 1080_balls_of_solitude.py_) with remote sever.But I can run the example in my personal computer.The diffrence of environment are the version of cuda and nvidia driver.Why?So confused.

LJoson, 2023-11-07:
> I also met this problem when I run the example(p_ython 1080_balls_of_solitude.py_) with remote sever.But I can run the example in my personal computer.The diffrence of environment are the version of cuda and nvidia driver.Why?So confused.

It is possible that the remote server does not have GUI remote services, and when using Isaac, the startup items that depend on GUI training are not closed (but I also haven't understood how to close GUI to to simulation and training).

kc-ustc, 2023-11-07:
> It is possible that the remote server does not have GUI remote services, and when using Isaac, the startup items that depend on GUI training are not closed (but I also haven't understood how to [...]

Thanks for your answer!But our  remote server does have a GUI installed.You mean that when I run example like  **_1080_balls_of_solitude.py_**, I must shut down the remote server GUI?

LJoson, 2023-11-07:
> Thanks for your answer!But our remote server does have a GUI installed.You mean that when I run example like **_1080_balls_of_solitude.py_**, I must shut down the remote server GUI?

emmm,i mean that when you tryrun example like 1080_balls_of_solitude.py,your server must have a GUI

LJoson, 2023-11-07:
> Thanks for your answer!But our remote server does have a GUI installed.You mean that when I run example like **_1080_balls_of_solitude.py_**, I must shut down the remote server GUI?

because  found that the issue was actually due to the lack of Vulkan installation(GUI underlying driver

kc-ustc, 2023-11-07:
> (GUI underlying drive

Thank you, I naively thought ubuntu20.04 came with Vulkan.

asibatian, 2024-06-25:
Here is my experience: I try to train a policy for legged gym and set `headless = False` to visualize the training process.
When waiting for loading the terrains into isaac gym, it throws `segmentation fault (core dumped)`, after waiting for about 1 minute.
But when I reduce the number of terrains, Isaac Gym load the terrains within 1 minute and it works fine.
In both case, my GPU memory is not full. 
So, I guess there is a time limits for loading terrain triangles.

I do not solve this problem.

formoree, 2024-09-05:
> > (GUI underlying drive （GUI 底层驱动器
> 
> Thank you, I naively thought ubuntu20.04 came with Vulkan.谢谢，我天真地以为 ubuntu20.04 是 Vulkan 附带的。

Have you ever solved this problem? I have install Vulkna, but the problem is still there.

MiangChen, 2024-11-21:
GPU: 4070 Ti super(release 2024/2）
at first, I install  `NVIDIA-Linux-x86_64-550.135` and `cuda_12.4.1_550.54.15_linux`, and when I run `python 1080_balls_of_solitude.py `, segmentation fault occurs

then, I simply downgrade the version: `NVIDIA-Linux-x86_64-535.171.04` and `cuda_12.2.2_535.104.05_linux`, then, this bug disapper

so, i simply suggest that you can try to downgrade the NVIDIA Driver and Cuda version.

yijionglin, 2025-07-01:
I can run with numEnvs= 2048 without any problem, but met this problem with numEnvs= 4096... Any thoughts to this guys?

nuomizai, 2025-07-02:
> With my customized environments, I can run with numEnvs= 2048 without any problem, but met this problem with numEnvs= 4096... Any thoughts to this guys?

It might be exceeding the GPU memory.