isaac-sim/IsaacGymEnvs #53: Failed to create SDF triangle mesh

quantumiracle, 2022-07-05:
Hi, 

I'm trying to use factory environments for training. Specifically, with `python train.py task=FactoryTaskNutBoltPick`. I receive the error `[Error] [carb.gym.plugin] Failed to create SDF triangle mesh`. And the full log of error report is as following:
```
Importing module 'gym_37' (/home/zihan/software/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_37.so)
Setting GYM_USD_PLUG_INFO_PATH to /home/zihan/software/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
train.py:49: UserWarning: 
The version_base parameter is not specified.
Please specify a compatability version level, or None.
Will assume defaults for version 1.1
  @hydra.main(config_name="config", config_path="./cfg")
/home/zihan/anaconda3/envs/x/lib/python3.7/site-packages/hydra/_internal/defaults_list.py:251: UserWarning: In 'config': Defaults list is missing `_self_`. See https://hydra.cc/docs/upgrades/1.0_to_1.1/default_composition_order for more information
  warnings.warn(msg, UserWarning)
/home/zihan/anaconda3/envs/x/lib/python3.7/site-packages/hydra/_internal/defaults_list.py:415: UserWarning: In config: Invalid overriding of hydra/job_logging:
Default list overrides requires 'override' keyword.
See https://hydra.cc/docs/next/upgrades/1.0_to_1.1/defaults_list_override for more information.

  deprecation_warning(msg)
/home/zihan/anaconda3/envs/x/lib/python3.7/site-packages/hydra/_internal/hydra.py:127: UserWarning: Future Hydra versions will no longer change working directory at job runtime by default.
See https://hydra.cc/docs/next/upgrades/1.1_to_1.2/changes_to_job_working_dir/ for more information.
  configure_logging=with_log_configuration,
PyTorch version 1.11.0.dev20220110
Device count 8
/home/zihan/software/isaacgym/python/isaacgym/_bindings/src/gymtorch
Using /home/zihan/.cache/torch_extensions/py37_cu113 as PyTorch extensions root...
Emitting ninja build file /home/zihan/.cache/torch_extensions/py37_cu113/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
Loading extension module gymtorch...
/home/zihan/software/isaacgym/python/isaacgym/torch_utils.py:135: DeprecationWarning: `np.float` is a deprecated alias for the builtin `float`. To silence this warning, use `float` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.float64` here.
Deprecated in NumPy 1.20; for more details and guidance: https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
  def get_axis_params(value, axis_idx, x_value=0., dtype=np.float, n_dims=3):
2022-07-05 14:30:06,887 - INFO - logger - logger initialized
<unknown>:6: DeprecationWarning: invalid escape sequence \*
Error: FBX library failed to load - importing FBX data will not succeed. Message: No module named 'fbx'
FBX tools must be installed from https://help.autodesk.com/view/FBX/2020/ENU/?guid=FBX_Developer_Help_scripting_with_python_fbx_installing_python_fbx_html
task: 
    name: FactoryTaskNutBoltPick
    physics_engine: physx
    sim: 
        use_gpu_pipeline: True
        up_axis: z
        dt: 0.016667
        gravity: [0.0, 0.0, -9.81]
        disable_gravity: False
    env: 
        numEnvs: 128
        numObservations: 20
        numActions: 12
        close_and_lift: True
        num_gripper_move_sim_steps: 20
        num_gripper_close_sim_steps: 25
        num_gripper_lift_sim_steps: 25
    randomize: 
        franka_arm_initial_dof_pos: [0.3413, -0.8011, -0.067, -1.8299, 0.0266, 1.0185, 1.0927]
        fingertip_midpoint_pos_initial: [0.0, -0.2, 0.2]
        fingertip_midpoint_pos_noise: [0.2, 0.2, 0.1]
        fingertip_midpoint_rot_initial: [3.1416, 0, 3.1416]
        fingertip_midpoint_rot_noise: [0.3, 0.3, 1]
        nut_pos_xy_initial: [0.0, -0.3]
        nut_pos_xy_initial_noise: [0.1, 0.1]
        bolt_pos_xy_initial: [0.0, 0.0]
        bolt_pos_xy_noise: [0.1, 0.1]
    rl: 
        pos_action_scale: [0.1, 0.1, 0.1]
        rot_action_scale: [0.1, 0.1, 0.1]
        force_action_scale: [1.0, 1.0, 1.0]
        torque_action_scale: [1.0, 1.0, 1.0]
        clamp_rot: True
[... 137 lines cut ...]
force_render: True
headless: True
Setting seed: 42
self.seed = 42
Started to train
Exact experiment name requested from command line: FactoryTaskNutBoltPick
/home/zihan/anaconda3/envs/x/lib/python3.7/site-packages/gym/spaces/box.py:112: UserWarning: WARN: Box bound precision lowered by casting to float32
  logger.warn(f"Box bound precision lowered by casting to {self.dtype}")
Not connected to PVD
+++ Using GPU PhysX
Physics Engine: PhysX
Physics Device: cuda:0
GPU Pipeline: enabled
Using SDF cache directory '/home/zihan/.isaacgym/sdf_V100'
~!~!~! Loaded/Cooked SDF triangle mesh 0 @ 0x55f2e8c6b7a0, resolution=256, spacing=0.000108
  ~!~! Bounds:  (-0.012000, 0.012000) (-0.013856, 0.013856) (0.016000, 0.029000)
  ~!~! Extents: (0.024000, 0.027712, 0.013000)
  ~!~! Resolution: (222, 256, 121)
[Error] [carb.gym.plugin] Failed to create SDF triangle mesh
Segmentation fault (core dumped)
```

Any idea of why this is happening?

Thanks!

kstorey-nvidia, 2022-07-06:
I'm not sure, but could it be that you have a corrupted cached file here:

Using SDF cache directory '/home/zihan/.isaacgym/sdf_V100'

Can you try clearing the cache directory and letting the test run again. It may take a while to run the first time. That's when it's generating the SDF. After the first run, it should start quickly next time. If you terminate while it's generating the SDF for the first run, I'm not sure exactly what state you might end up in, but it's possible that the file has already been created but it doesn't contain any data.

ynarang, 2022-07-26:
@quantumiracle  Did the solution above work for you? We had an internal user of Factory who ran into the same issue, and it was resolved by clearing the SDF cache directory, relaunching the Factory task, and waiting until SDF generation was complete.

quantumiracle, 2022-07-26:
Hi, it seems the problem is solved, but I'm not quite sure how. I probably reinstalled it and it seems to work now.

closed 2022-07-26 by quantumiracle

Bdl-1989, 2023-12-19:
I meet the same issue only run torchrun directly.
But I can solve by run the python train.py first to create SDF file and then torchrun