isaac-sim/IsaacGymEnvs #54: [Error] [carb.gym.plugin] *** Failed to load './/../assets/glb/ingenuity/chassis.glb'!

Toni-SM, 2022-07-11:
Hi

Importing the **Ingenuity** task/environment outside the `IsaacGymEnvs/isaacgymenvs` folder produces the following error:

```
(env_isaacgym4) toni@HP-ZBook-Studio-G8:~/Documents/examples/isaacgym$ python
Python 3.8.10 (default, Mar 15 2022, 12:22:08) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import isaacgym
Importing module 'gym_38' (/home/toni/isaacgym/isaacgym4/python/isaacgym/_bindings/linux-x86_64/gym_38.so)
Setting GYM_USD_PLUG_INFO_PATH to /home/toni/isaacgym/isaacgym4/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
>>> import isaacgymenvs
>>> env = isaacgymenvs.make(seed=42, 
...                         task="Ingenuity", 
...                         num_envs=4096, 
...                         sim_device="cuda:0",
...                         rl_device="cuda:0",
...                         graphics_device_id=0,
...                         headless=False)
PyTorch version 1.12.0+cu116
Device count 1
/home/toni/isaacgym/isaacgym4/python/isaacgym/_bindings/src/gymtorch
Using /home/toni/.cache/torch_extensions/py38_cu116 as PyTorch extensions root...
Emitting ninja build file /home/toni/.cache/torch_extensions/py38_cu116/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
Loading extension module gymtorch...
/home/toni/isaacgym/isaacgym4/python/isaacgym/torch_utils.py:135: DeprecationWarning: `np.float` is a deprecated alias for the builtin `float`. To silence this warning, use `float` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.float64` here.
Deprecated in NumPy 1.20; for more details and guidance: https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
  def get_axis_params(value, axis_idx, x_value=0., dtype=np.float, n_dims=3):
2022-07-11 17:25:28,797 - INFO - logger - logger initialized
<unknown>:3: DeprecationWarning: invalid escape sequence \*
Error: FBX library failed to load - importing FBX data will not succeed. Message: No module named 'fbx'
FBX tools must be installed from https://help.autodesk.com/view/FBX/2020/ENU/?guid=FBX_Developer_Help_scripting_with_python_fbx_installing_python_fbx_html
/home/toni/isaacgym/isaacgym4/IsaacGymEnvs/isaacgymenvs/__init__.py:35: UserWarning: 
The version_base parameter is not specified.
Please specify a compatability version level, or None.
Will assume defaults for version 1.1
  with initialize(config_path="./cfg"):
/home/toni/Documents/envs/env_isaacgym4/lib/python3.8/site-packages/hydra/_internal/defaults_list.py:251: UserWarning: In 'config': Defaults list is missing `_self_`. See https://hydra.cc/docs/upgrades/1.0_to_1.1/default_composition_order for more information
  warnings.warn(msg, UserWarning)
/home/toni/Documents/envs/env_isaacgym4/lib/python3.8/site-packages/hydra/_internal/defaults_list.py:415: UserWarning: In config: Invalid overriding of hydra/job_logging:
Default list overrides requires 'override' keyword.
See https://hydra.cc/docs/next/upgrades/1.0_to_1.1/defaults_list_override for more information.

  deprecation_warning(msg)
/home/toni/Documents/envs/env_isaacgym4/lib/python3.8/site-packages/gym/spaces/box.py:112: UserWarning: WARN: Box bound precision lowered by casting to float32
  logger.warn(f"Box bound precision lowered by casting to {self.dtype}")
[Warning] [carb.gym.plugin] useGpu is set, forcing single scene (0 subscenes)
Not connected to PVD
+++ Using GPU PhysX
Physics Engine: PhysX
Physics Device: cuda:0
GPU Pipeline: enabled
MESA-INTEL: warning: Performance support disabled, consider sysctl dev.i915.perf_stream_paranoid=0

[Error] [carb.gym.plugin] *** Failed to load './/../assets/glb/ingenuity/chassis.glb'!

Segmentation fault (core dumped)
```
<hr>

Problem: The path to the assets is relative only to the `isaacgymenvs` python package.

Steps to reproduce:
1. Open python (interactive terminal) in any folder except `IsaacGymEnvs/isaacgymenvs`
2. Execute the following code
  ```python
  import isaacgym
  import isaacgymenvs
  env = isaacgymenvs.make(seed=42, 
                        task="Ingenuity", 
                        num_envs=4096, 
                        sim_device="cuda:0",
                        rl_device="cuda:0",
                        graphics_device_id=0,
                        headless=False)
  ```

Toni-SM, 2023-01-04:
Hi @dulucas 

Nop...

Each time I want to play with this task, I need to copy the main script to the `IsaacGymEnvs/isaacgymenvs` folder and run it from there

HiuhiuWong, 2023-12-07:
I have the same errors:
python test.py 
Importing module 'gym_37' (/home/felicityw/ISSAC/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_37.so)
Setting GYM_USD_PLUG_INFO_PATH to /home/felicityw/ISSAC/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
/home/felicityw/miniconda3/envs/rlgpu/lib/python3.7/site-packages/torch/utils/cpp_extension.py:3: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import imp
/home/felicityw/miniconda3/envs/rlgpu/lib/python3.7/site-packages/torch/utils/cpp_extension.py:23: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  from pkg_resources import packaging  # type: ignore
/home/felicityw/miniconda3/envs/rlgpu/lib/python3.7/site-packages/pkg_resources/__init__.py:2871: DeprecationWarning: Deprecated call to `pkg_resources.declare_namespace('google')`.
Implementing implicit namespace packages (as specified in PEP 420) is preferred to `pkg_resources.declare_namespace`. See https://setuptools.pypa.io/en/latest/references/keywords.html#keyword-namespace-packages
  declare_namespace(pkg)
PyTorch version 1.8.1
Device count 1
/home/felicityw/ISSAC/isaacgym/python/isaacgym/_bindings/src/gymtorch
Using /home/felicityw/.cache/torch_extensions as PyTorch extensions root...
Emitting ninja build file /home/felicityw/.cache/torch_extensions/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
Loading extension module gymtorch...
2023-12-07 12:09:16,568 - INFO - logger - logger initialized
<unknown>:6: DeprecationWarning: invalid escape sequence \*
Error: FBX library failed to load - importing FBX data will not succeed. Message: No module named 'fbx'
FBX tools must be installed from https://help.autodesk.com/view/FBX/2020/ENU/?guid=FBX_Developer_Help_scripting_with_python_fbx_installing_python_fbx_html
/home/felicityw/ISSAC/isaacgym/python/isaacgym/torch_utils.py:135: DeprecationWarning: `np.float` is a deprecated alias for the builtin `float`. To silence this warning, use `float` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.float64` here.
Deprecated in NumPy 1.20; for more details and guidance: https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
  def get_axis_params(value, axis_idx, x_value=0., dtype=np.float, n_dims=3):
/home/felicityw/miniconda3/envs/rlgpu/lib/python3.7/site-packages/networkx/classes/graph.py:23: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated since Python 3.3,and in 3.9 it will stop working
  from collections import Mapping
/home/felicityw/miniconda3/envs/rlgpu/lib/python3.7/site-packages/networkx/classes/reportviews.py:95: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated since Python 3.3,and in 3.9 it will stop working
  from collections import Mapping, Set, Iterable
/home/felicityw/miniconda3/envs/rlgpu/lib/python3.7/site-packages/networkx/readwrite/graphml.py:346: DeprecationWarning: `np.int` is a deprecated alias for the builtin `int`. To silence this warning, use `int` by itself. Doing this will not modify any behavior and is safe. When replacing `np.int`, you may wish to use e.g. `np.int64` or `np.int32` to specify the precision. If you wish to review your current use, check the release note link for additional information.
Deprecated in NumPy 1.20; for more details and guidance: https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
  (np.int, "int"), (np.int8, "int"),
/home/felicityw/pyeg/IsaacGymEnvs/isaacgymenvs/__init__.py:35: UserWarning: 
The version_base parameter is not specified.
Please specify a compatability version level, or None.
Will assume defaults for version 1.1
  with initialize(config_path="./cfg"):
/home/felicityw/miniconda3/envs/rlgpu/lib/python3.7/site-packages/gym/spaces/box.py:84: UserWarning: WARN: Box bound precision lowered by casting to float32
  logger.warn(f"Box bound precision lowered by casting to {self.dtype}")
[Warning] [carb.gym.plugin] useGpu is set, forcing single scene (0 subscenes)
Not connected to PVD
+++ Using GPU PhysX
Physics Engine: PhysX
Physics Device: cuda:0
GPU Pipeline: enabled
[Error] [carb.gym.plugin] *** Failed to load './/../assets/glb/ingenuity/chassis.glb'!

段错误 (核心已转储)