isaac-sim/IsaacGymEnvs #56: Visualizing Contact Force

Kashu7100, 2022-07-12:
Hello, 

Is there any way to render the (foot) contact force as an arrow in the simulator?
Any code snippet is appreciated. 

Sincerely,

jinseokbae, 2022-07-13:
You can visualize contacts when running simulation in the CPU device.
Try using `draw_env_soft_contacts`.

Kashu7100, 2022-07-13:
Thank you for your reply. 
Is there any way to do this on GPU (I got the following error when trying on GPU)?

```bash 
[Error] [carb.gym.plugin] Function GymDrawEnvRigidContacts cannot be used with the GPU pipeline after simulation starts.  Please use the tensor API if possible.  See docs/programming/tensors.html for more info.
```

jinseokbae, 2022-07-13:
I've tried only in the CPU pipeline.
Maybe it is yet to be implemented in GPU pipeline.
I think you need to wait for the developer's reply.
Hope figuring this out!

closed 2022-07-19 by Kashu7100

Yasu31, 2023-02-16:
I would like to add that in order for the contacts to be processed, the `contact_collection` parameter for physx must be set to `CC_LAST_SUBSTEP` or `CC_ALL_SUBSTEPS` (though the latter says "broken - do not use!" so the former is probably better) instead of CC_NEVER, which is the default for some environments, e.g. https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/blob/dee7c56765e14f6f4344c4d2e91d7a9eb3bfa619/isaacgymenvs/cfg/task/ShadowHand.yaml#L190