isaac-sim/IsaacGymEnvs #59: How to determine geometry size of peg and hole

quantumiracle, 2022-07-14:
Hi, 

I notice that in factory nut-and-bolt environment, the specific width and height of nuts and bolts are given in yaml file as [here](https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/blob/6da538b44618afb9086be30371e5466ea182c2bf/assets/factory/yaml/factory_asset_info_nut_bolt.yaml#L4). However, for insertion tasks, there is no such information, as [here](https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/blob/6da538b44618afb9086be30371e5466ea182c2bf/assets/factory/yaml/factory_asset_info_insertion.yaml#L4). May I task is it possible to get these info?

Thanks

ynarang, 2022-07-26:
@quantumiracle Sorry about the delay. Absolutely, we can add this information to the YAML file. We'll get this merged in shortly.

Also note that we will soon be releasing Onshape CAD files as well, which will allow you to make any additional measurements you might want.

quantumiracle, 2022-07-26:
It's great to hear that! Waiting for it..

ynarang, 2022-07-29:
@quantumiracle I've made the requested additions to the YAML file, along with a few other revisions. They should be pushed to the public repo within the next few days.

ynarang, 2022-08-15:
@quantumiracle Sorry for the delay on this, as we combined these updates with recent updates on domain randomization. (For reference, please see the [release notes](https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/commit/ca7a4fb762f9581e39cc2aab644f18a83d6ab0ba).) Let us know if there's any further information you need for your work.

closed 2022-08-15 by gavrielstate