isaac-sim/IsaacGymEnvs #62: `libmem_filesys.so: cannot open shared object file`

vwxyzjn, 2022-07-19:
Running into the issue of `libmem_filesys.so: cannot open shared object file`. I tried googling but could not find any info on this file, which is also not shipped with Preview 4. 

[image]

Additionally, any chance the Preview 3 and prior versions can be downloaded somewhere? Being able to download them might help with reproducibility :) 

```bash
python joint_monkey.py 
Importing module 'gym_37' (/home/costa/Documents/go/src/github.com/cleanrl/cleanrl/ppo_continuous_action_isaacgym/isaacgym/isaacgym/_bindings/linux-x86_64/gym_37.so)
[Error] [carb] [Plugin: libcarb.gym.plugin.so] Could not load the dynamic library from /home/costa/Documents/go/src/github.com/cleanrl/cleanrl/ppo_continuous_action_isaacgym/isaacgym/isaacgym/_bindings/linux-x86_64/libcarb.gym.plugin.so. Error: libmem_filesys.so: cannot open shared object file: No such file or directory
[Warning] [carb] Potential plugin preload failed: /home/costa/Documents/go/src/github.com/cleanrl/cleanrl/ppo_continuous_action_isaacgym/isaacgym/isaacgym/_bindings/linux-x86_64/libcarb.gym.plugin.so
Setting GYM_USD_PLUG_INFO_PATH to /home/costa/Documents/go/src/github.com/cleanrl/cleanrl/ppo_continuous_action_isaacgym/isaacgym/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
[Error] [carb] Failed to acquire interface: [carb::gym::Gym v0.1], by client: carb.gym.python.gym_37 (plugin name: (null))
Traceback (most recent call last):
  File "joint_monkey.py", line 59, in <module>
    gym = gymapi.acquire_gym()
RuntimeError: Failed to acquire interface: carb::gym::Gym (pluginName: nullptr)
```

vwxyzjn, 2022-07-19:
Think I found it to be shipped with the `isaacgym` package. Gonna debug a bit more.

vwxyzjn, 2022-07-19:
Looks like this was an issue with my particular setup. Sorry for the ping.

gavrielstate, 2022-07-19:
@vwxyzjn  - might be useful for others to say what your issue was in case others come across it.

vwxyzjn, 2022-07-22:
Hey, sorry @gavrielstate, this was specific to my poetry setup. Regular users shouldn't encounter this issue. I have ironed out all the details of running IsaacGym w/ CleanRL. Would you be interested in giving it a try?

The PR is at https://github.com/vwxyzjn/cleanrl/pull/233, and the installation instruction is at https://cleanrl-git-isaacgym-vwxyzjn.vercel.app/rl-algorithms/ppo/#usage_8

Should be as straightforward as 

```
git clone https://github.com/vwxyzjn/cleanrl.git
git checkout isaacgym

cd cleanrl/ppo_continuous_action_isaacgym
# put `IsaacGym_Preview_4_Package.tar.gz` here 
stat IsaacGym_Preview_4_Package.tar.gz
mkdir temp_isaacgym
tar -xf IsaacGym_Preview_4_Package.tar.gz -C temp_isaacgym
mv temp_isaacgym/isaacgym/python/* isaacgym
rm -rf temp_isaacgym

poetry install -E isaacgym
python ppo_continuous_action_isaacgym.py --help
python ppo_continuous_action_isaacgym.py --env-id Ant
```

vwxyzjn, 2022-09-27:
Leave a note here. The cause of the issue seems to be that the `isaacgym` package is not setup as a usual python package, so it is apparently really important to do an **editable** install like `pip install -e isaacgym`, otherwise this error would appear. 

The current distribution mode of Isaac Gym feels quite tricky, and it could make things challenging to reproduce. @gavrielstate, would it be possible to revisit #23 and consider an API-KEY-based approach, where the user would need to agree to the license agreement to get a key to run the local Isaac gym?

scharalambous3, 2023-05-07:
Hi @vwxyzjn,
Thanks for sharing this info. Have you been able to understand the root of the problem?
In my case the same erros
> *** Warning: failed to preload USD libs
and
> Error: libmem_filesys.so: cannot open shared object file: No such file or directory
arise when I try to load a pybind11 wrapper of a C++ package
This error persists even when installing them with the `-e` flag
Any help will be greatly appreciated!

vwxyzjn, 2023-05-07:
Could you give this a try: https://docs.cleanrl.dev/rl-algorithms/ppo/#usage_10. 

I just gave it a try and it still works. 
[image]

scharalambous3, 2023-05-07:
I'm not using cleanrl, but I've encountered the same error from a different setup (trying to import a pybind11 wrapped C++ package).Have you been able to identify what environment variable was the root of the problem? For example, sys.path doesn't seem to be it

DanHrmti, 2023-07-17:
@scharalambous3 have you been able to find a solution for this?
I have encountered the same warning+Error when trying to run the `joint_monkey.py` example on a clean installation of isaacgym.

scharalambous3, 2023-07-17:
The source of the error in my case was a conflict in my environment. The libtf.so of isaacgym was conflicting with the respective one of ROS. Hope that helps

DanHrmti, 2023-07-19:
Not exactly my case, but thank you for the quick reply :)

RayYoh, 2023-09-15:
> Not exactly my case, but thank you for the quick reply :)

How did you tickle it?

DanHrmti, 2023-09-17:
Adding the path to the library's location (...../isaacgym/python/isaacgym/_bindings/linux-x86_64) to the environment variable LD_LIBRARY_PATH solved the `"Error: libmem_filesys.so: cannot open shared object file: No such file or directory"` error for me.

`"*** Warning: failed to preload USD libs"` error message persists but does not prevent IsaacGym from working properly in my case.

LouieTsai71, 2024-02-27:
> Adding the path to the library's location (...../isaacgym/python/isaacgym/_bindings/linux-x86_64) to the LD_LIBRARY_PATH environment variable solved the `"Error: libmem_filesys.so: cannot open shared object file: No such file or directory"` error for me.
> 
> `"*** Warning: failed to preload USD libs"` error message persists but does not prevent IsaacGym from working properly in my case.

May I ask how you did this exactly?

DanHrmti, 2024-03-04:
I added the following line to the end of the `.bashrc` file: `export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:<PATH>`,
where `<PATH>` is the path to the relevant library's location.

HariP19, 2024-08-17:
> The source of the error in my case was a conflict in my environment. The libtf.so of isaacgym was conflicting with the respective one of ROS. Hope that helps

@scharalambous3 Do you remember how you managed to solve the libtf.so issue? I am trying to use a  ros package with isaacgym using pybind11 and I am facing the same issue. Importing the ROS lib first causes undefined symbol errors in importing Isaac Gym and vice versa

scharalambous3, 2024-08-17:
There is a name conflict with one of isaac's libraries. You must rebuild the ROS tf library from source and rename it to something else