isaac-sim/IsaacGymEnvs #63: Are the limitations regarding mass and scale randomization still true?

johannespitz, 2022-07-19:
https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/blob/6da538b44618afb9086be30371e5466ea182c2bf/docs/domain_randomization.md?plain=1#L99
In the documentation under "Programming/Physics" I can only find this note:
```
Actor scaling has known limitations when using the GPU pipeline. The center-of-mass of rigid bodies is not updated, which only affects bodies with center-of-mass not at the body origin. Also, resetting transforms and velocities or applying forces in the same simulation step as scaling may yield incorrect results. We expect these limitation to be resolved in the future.
```
Does that mean I can freely randomize masses and even the scale (if the center of mass of in the origin) during `reset()`?

gavrielstate, 2022-07-19:
You should be able to randomize them during reset, but you will lose determinism. See here: https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/blob/main/docs/reproducibility.md

johannespitz, 2022-07-19:
Thank you for the quick response.
I'll keep an eye on the reproducibility.

closed 2022-07-19 by johannespitz