isaac-sim/IsaacGymEnvs #71: import ERROR

junghs1040, 2022-08-06:
Hi i got an error 
i want to try task=Cartpole, but the following error occured
`Traceback (most recent call last):
  File "/home/seok2/anaconda3/envs/rlstudy/lib/python3.8/site-packages/hydra/_internal/utils.py", line 252, in run_and_report
    assert mdl is not None
AssertionError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "train.py", line 188, in <module>
    launch_rlg_hydra()
  File "/home/seok2/anaconda3/envs/rlstudy/lib/python3.8/site-packages/hydra/main.py", line 48, in decorated_main
    _run_hydra(
  File "/home/seok2/anaconda3/envs/rlstudy/lib/python3.8/site-packages/hydra/_internal/utils.py", line 377, in _run_hydra
    run_and_report(
  File "/home/seok2/anaconda3/envs/rlstudy/lib/python3.8/site-packages/hydra/_internal/utils.py", line 294, in run_and_report
    raise ex
  File "/home/seok2/anaconda3/envs/rlstudy/lib/python3.8/site-packages/hydra/_internal/utils.py", line 211, in run_and_report
    return func()
  File "/home/seok2/anaconda3/envs/rlstudy/lib/python3.8/site-packages/hydra/_internal/utils.py", line 378, in <lambda>
    lambda: hydra.run(
  File "/home/seok2/anaconda3/envs/rlstudy/lib/python3.8/site-packages/hydra/_internal/hydra.py", line 111, in run
    _ = ret.return_value
  File "/home/seok2/anaconda3/envs/rlstudy/lib/python3.8/site-packages/hydra/core/utils.py", line 233, in return_value
    raise self._return_value
  File "/home/seok2/anaconda3/envs/rlstudy/lib/python3.8/site-packages/hydra/core/utils.py", line 160, in run_job
    ret.return_value = task_function(task_cfg)
  File "train.py", line 51, in launch_rlg_hydra
    from isaacgymenvs.utils.rlgames_utils import RLGPUEnv, RLGPUAlgoObserver, get_rlgames_env_creator
  File "/home/seok2/IsaacGymEnvs/isaacgymenvs/utils/rlgames_utils.py", line 36, in <module>
    from tasks import isaacgym_task_map
  File "/home/seok2/isaacgym/python/IsaacGymEnvs/isaacgymenvs/tasks/__init__.py", line 30, in <module>
    from .ant import Ant
  File "/home/seok2/isaacgym/python/IsaacGymEnvs/isaacgymenvs/tasks/ant.py", line 38, in <module>
    from isaacgymenvs.tasks.base.vec_task import VecTask
  File "/home/seok2/IsaacGymEnvs/isaacgymenvs/tasks/__init__.py", line 31, in <module>
    from tasks.ant import Ant
ImportError: cannot import name 'Ant' from partially initialized module 'tasks.ant' (most likely due to a circular import) (/home/seok2/isaacgym/python/IsaacGymEnvs/isaacgymenvs/tasks/ant.py)
`

ChengEeee, 2022-08-11:
I have the same issue.Have you solved it？

junghs1040, 2022-08-12:
> I have the same issue.Have you solved it？

Yes i solved 
I was using conda environment and i did pip install - e in base environment

I did pip install -e again in right conda environment again and it worked!

closed 2022-08-18 by junghs1040