isaac-sim/IsaacGymEnvs #73: FrankaCubeStack not converging

MohammadYasar, 2022-08-11:
Hi, I have been training FrankaCubeStack with the default parameters in cfg/task/FrankaCubeStack.yaml and cfg/train/FrankaCubeStackPPO.yaml. The reward stops increasing after approx. 800 and 10,000 epochs. I have also changed the parameter values for minibatch_size to 131072 and numEnvs to 16384 and kl_threshold to 1.6e-2, as reported in the paper, but haven't been able to get the results as reported in the paper. 
Here's the train command:
`python train.py headless=True task=FrankaCubeStack wandb_activate=True wandb_entity=nvidia wandb_project=FrankaCubeStack`

[image]

Please let me know if there's something I'm doing wrong.

DanielTakeshi, 2022-09-10:
I can reproduce the result from @MohammadYasar . Here is the curve I see on wandb:

[image]

I ran this using:

```
python train.py task=FrankaCubeStack headless=True wandb_activate=True wandb_entity=A wandb_project=B
```

where I replaced `A` with my wandb username and `B` with the project name I used. 

This is with the main branch and the latest commit as of today: https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/commit/ca7a4fb762f9581e39cc2aab644f18a83d6ab0ba

I did not change any of the PPO settings.

Also, it took about 264 minutes to run which is a bit longer than expected; for reference, running the above but with `FrankaCabinet` took about 20 minutes.

DanielTakeshi, 2022-09-10:
Quick follow-up, I decided to load the policy rollouts and visualize them:

https://youtu.be/ACwcYy3Xb08

From the video it actually seems like the Franka is learning to do Cube stacking? I think it's succeeding most of the time.

So maybe the reward scale is different now as compared to the Isaac Gym white paper as reported in https://arxiv.org/abs/2108.10470 ?

MohammadYasar, 2022-09-30:
I think it should stack and move away from the cube. I was able to reproduce the graph by removing the `stack_reward>0 ` from Line 746:
 `reset_buf = torch.where((progress_buf >= max_episode_length - 1) , torch.ones_like(reset_buf), reset_buf)
` 
[image]

closed 2022-10-02 by MohammadYasar