isaac-sim/IsaacGymEnvs #75: RecordVideo Class Problem in Gym.wrapper

Touutae-lab, 2022-08-13:
here is my code 

```
envs = isaacgymenvs.make(
	seed=0, 
	task="Ant", 
	num_envs=20, 
	sim_device="cuda:0",
	rl_device="cuda:0",
	graphics_device_id=0,
	headless=False,
	multi_gpu=False,
	virtual_screen_capture=True,
	force_render=False,
)
envs.is_vector_env = True
envs = gym.wrappers.RecordVideo(
    env=envs,
    episode_trigger=lambda step: step % 10 == 0,
    video_folder="Videos",
    video_length=100,
)
envs.reset() << Bug this line
```
**'Ant' object has no attribute 'render_mode'**

younik, 2022-08-15:
This is happening because of the new Render API (https://github.com/openai/gym/pull/2671).
I suggest downgrading your gym version until the environment will follow the new convention

closed 2022-08-17 by Touutae-lab

Touutae-lab, 2022-08-17:
for now who using this i suggest to use gym 0.24 instead of 0.25