isaac-sim/IsaacGymEnvs #76: Cannot resolve mesh for bnc, usb, dsub

quantumiracle, 2022-08-23:
Hi, 

I'm working on FactoryTaskInsertion, which runs successfully with round or rectangular type objects. But for bnc, usb, dsub, it report similar errors in the mesh creating process (take usb as an example):
```
  logger.warn('Please be patient: SDFs may be generating, which may take a few minutes. Terminating prematurely may result in a corrupted SDF cache.')
[Error] [carb.gym.plugin] Failed to resolve collision mesh '../mesh/factory_insertion/factory_usb_plug_coll.obj'
[Error] [carb.gym.plugin] Failed to resolve visual mesh '../mesh/factory_insertion/factory_usb_plug_viz.obj'
[Error] [carb.gym.plugin] Failed to resolve collision mesh '../mesh/factory_insertion/factory_usb_socket_coll.obj'
[Error] [carb.gym.plugin] Failed to resolve visual mesh '../mesh/factory_insertion/factory_usb_socket_viz.obj'
Error executing job with overrides: ['task=FactoryTaskInsertion']
```

ynarang, 2022-09-13:
@quantumiracle, great to hear that you are successfully training on FactoryTaskInsertion, and very sorry about the delayed response! 

The reason you are seeing those errors is that we have not yet released the OBJ files for those parts. We sourced high-quality CAD models from manufacturers and then modified them manually in CAD software to simplify external features, occasionally simplify internal features (e.g., if there were internal elements that required deformable-body simulation, which Gym does not currently expose from PhysX 5.1), and exactly preserve most contact geometry. We have to get approval from the manufacturers to make sure that we can release our modified versions.

In order to unblock you, we'd like to provide the links to the websites from which the original models were obtained, along with some others that were not mentioned in the Factory paper but are based on the NIST assembly benchmark. Look for the CAD link on each page. You can open these files in CAD software, and with some CAD experience, create models very similar to the ones we developed for Factory. Afterward, you can export the CAD models to OBJ for use in Gym.

BNC plug ([item](https://www.digikey.com/en/products/detail/amphenol-rf/112420/1989856))
BNC socket ([item](https://www.digikey.com/en/products/detail/molex/0731010120/1465130))
D-sub plug ([item](https://www.digikey.com/en/products/detail/assmann-wsw-components/A-DSF-25LPIII-Z/924268))
D-sub socket ([item](https://www.digikey.com/en/products/detail/assmann-wsw-components/A-DFF-25LPIII-Z/924259))
RJ45 plug ([item](https://www.digikey.com/en/products/detail/harting/09454521509/3974500))
RJ45 socket ([item](https://www.digikey.com/en/products/detail/amphenol-icc-fci/54602-908LF/1001360))
USB plug ([item](https://www.digikey.com/en/products/detail/bulgin/PX0441-2M00/1625994))
USB socket ([item](https://www.digikey.com/en/products/detail/amphenol-icc-fci/87520-0010BLF/1001359))
Waterproof plug ([item](https://b2b.harting.com/ebusiness/en_us/Han-High-Temp-10E-c-Male/09338102604))
Waterproof socket ([item](https://b2b.harting.com/ebusiness/en_us/Han-High-Temp-10E-c-Female/09338102704))

Hope this helps!

quantumiracle, 2022-09-13:
Understandable! Thanks for your response and providing these links. 
I will look into these.

closed 2022-09-13 by gavrielstate