isaac-sim/IsaacGymEnvs #78: `Capture videos during training` doesn't work

frt03, 2022-09-05:
Hi,

I tried `Capture videos during training` in README.md; i.e. running

```bash
python train.py capture_video=True capture_video_freq=1500 capture_video_len=100 force_render=False
```
but got the following error:

```
Traceback (most recent call last):
  File "train.py", line 156, in launch_rlg_hydra
    runner.run({
  File "/opt/conda/lib/python3.8/site-packages/rl_games/torch_runner.py", line 120, in run
    self.run_train(args)
  File "/opt/conda/lib/python3.8/site-packages/rl_games/torch_runner.py", line 101, in run_train
    agent.train()
  File "/opt/conda/lib/python3.8/site-packages/rl_games/common/a2c_common.py", line 1162, in train
    self.obs = self.env_reset()
  File "/opt/conda/lib/python3.8/site-packages/rl_games/common/a2c_common.py", line 470, in env_reset
    obs = self.vec_env.reset()
  File "/home/root/furuta/IsaacGymEnvs/isaacgymenvs/utils/rlgames_utils.py", line 159, in reset
    return self.env.reset()
  File "/opt/conda/lib/python3.8/site-packages/gym/wrappers/record_video.py", line 109, in reset
    self.start_video_recorder()
  File "/opt/conda/lib/python3.8/site-packages/gym/wrappers/record_video.py", line 121, in start_video_recorder
    self.video_recorder = video_recorder.VideoRecorder(
  File "/opt/conda/lib/python3.8/site-packages/gym/wrappers/monitoring/video_recorder.py", line 61, in __init__
    self.render_mode = env.render_mode
AttributeError: 'Ant' object has no attribute 'render_mode'

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.
Exception ignored in: <function VideoRecorder.__del__ at 0x7f1372356ca0>
Traceback (most recent call last):
  File "/opt/conda/lib/python3.8/site-packages/gym/wrappers/monitoring/video_recorder.py", line 269, in __del__
    self.close()
  File "/opt/conda/lib/python3.8/site-packages/gym/wrappers/monitoring/video_recorder.py", line 225, in close
    self.env.close()
AttributeError: 'Ant' object has no attribute 'close'
```

I guess it was caused by the API change of Open AI Gym. Can we fix this somehow? or could you add the valid version of gym in setup.py? FYI, I used `gym==0.25.2`.

Thank you for your great effort!

Touutae-lab, 2022-09-06:
Maybe this is the answer you were looking for https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/issues/75

Touutae-lab, 2022-09-06:
For now i suggest you to use `gym==0.24` instead of `gym==0.25`.

> Maybe this is the answer you were looking for #75

frt03, 2022-09-06:
@Touutae-lab Thanks!

For your reference, I tried `gym==0.25.1` and `gym==0.24.1`. `gym==0.25.1` raised another API issue and `gym==0.24.1` showed some warnings (but it seemed to work). So, I'm using `gym==0.23.1`now.

closed 2022-09-06 by frt03