isaac-sim/IsaacGymEnvs #81: Where do the stiffness and damping values come from?

DanielTakeshi, 2022-09-22:
I am testing with the `FrankaCabinet` environment, and I see that there are several values that seem to be hard-coded for stiffness and damping, in particular here:

https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/blob/ca7a4fb762f9581e39cc2aab644f18a83d6ab0ba/isaacgymenvs/tasks/franka_cabinet.py#L162-L163

and here:

https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/blob/ca7a4fb762f9581e39cc2aab644f18a83d6ab0ba/isaacgymenvs/tasks/franka_cabinet.py#L179-L201

My question: how does one determine the stiffness and damping values?  Or to make it more precise, how did these values come to be in the existing code?

The documentation in the "physics" section mentions stiffness and damping, for example:

[image]

However the explanation is mostly limited to giving a few examples of values. I am wondering how to actually select (and tune) such values.

DanielTakeshi, 2022-11-28:
Hi all, just wondering if there was any advice on identifying what values to fill in for the stiffness and damping?

lorenmt, 2023-03-05:
Wondering too. These values also seem to be different from different example files...

DanielTakeshi, 2023-03-05:
@lorenmt I am still also unsure about this, also from this:
https://github.com/NVIDIA-Omniverse/OmniIsaacGymEnvs/issues/14

I think NVIDIA might be shifting to IsaacSim so it may be the case that we should shift efforts to understanding IsaacSim? (I'm not entirely sure ...)

lorenmt, 2023-03-05:
I didn't realise there was a gym in Isaac Sim as well. But I do know they recently release another Gym-like library called Orbit which is built on top of Isaac Sim: https://github.com/NVIDIA-Omniverse/Orbit. These libraries are so confusing. 

I thought the Isaac Gym was built to ease the research, removing all of the complicated rendering stuff and just focusing on clean APIs for state-based rendering? But I am not sure any more...