isaac-sim/IsaacGymEnvs #95: Error creating viewer: Sim->Graphics is nullptr

ustchlsong, 2022-11-22:
How can I get image obs from isaacgym in a headless device?
even if when I use a HEADED device , and set `headless = False, virtual_screen_capture =True`,  I tried following code.
```
import isaacgym
import isaacgymenvs
import torch

envs = isaacgymenvs.make(
    seed=0, 
    task="Ant", 
    num_envs=20, 
    sim_device="cuda:0",
    rl_device="cuda:0",
    headless=False, 
    virtual_screen_capture=True, 
)
print("Observation space is", envs.observation_space)
print("Action space is", envs.action_space)
obs = envs.reset()
img = envs.render()
for _ in range(20):
    obs, reward, done, info = envs.step(
        torch.rand((2000,)+envs.action_space.shape, device="cuda:0")
    )
```
GOT OUTPUT WITH ERROR:`[Error] [carb.gym.plugin] Error creating viewer: Sim->Graphics is nullptr`
It seems that the error raised by 235 line in vec_task.py
`            self.viewer = self.gym.create_viewer(
                self.sim, gymapi.CameraProperties())`
I can't find a solution to it , please help me.

ustchlsong, 2022-11-22:
What's more. Is there some methods I can get a images obs from the env. And is it possible for isaacgymenv to train a RL agent with a image input

xizaoqu, 2023-02-26:
> I've found the solution for your first question. `graphics_device_id` need to be passed to `isaacgymenvs.make`, otherwise it will be set to -1 by default and cause the error. I use the following code and it works well.
> 
> ```
> envs = isaacgymenvs.make(
>     seed=0,
>     task="Ant",
>     num_envs=2000,
>     sim_device="cuda:0",
>     rl_device="cuda:0",
>     graphics_device_id=0,
>     headless=False
> )
> ```

Thank you, ashuai.

wangjunyi9999, 2023-03-25:
> I've found the solution for your first question. `graphics_device_id` need to be passed to `isaacgymenvs.make`, otherwise it will be set to -1 by default and cause the error. I use the following [...]

Thanks, it works fine for me.  Or set headless as True to close the render window, but it seems not a good idea.

zhuoqun-chen, 2023-09-25:
> Thanks, it works fine for me. Or set headless as True to close the render window, but it seems not a good idea.

Thank you! Setting either to `graphics_device_id=0` or `headless=True` solves my problem.