isaac-sim/IsaacGymEnvs #105: Setting shape properties has no effect

johannespitz, 2023-01-18:
It looks like only about half of the available shape properties can be set while the others remain zero no matter what values I attempted to set. See the following script for more details.
This is probably a Isaac gym problem (not necessary IsaacGymEnvs), but I was hoping to get some feedback here. There is already a  forum post (unanswered) about this problem (https://forums.developer.nvidia.com/t/setting-rigit-shape-properties-doesnt-work/202094).
```  
from isaacgym import gymapi

# initialize gym
gym = gymapi.acquire_gym()

# configure sim
sim_params = gymapi.SimParams()
device_id = 0
sim = gym.create_sim(device_id, device_id, gymapi.SIM_PHYSX, sim_params)
assert sim is not None
viewer = gym.create_viewer(sim, gymapi.CameraProperties())
assert viewer is not None

# add ground plane
plane_params = gymapi.PlaneParams()
plane_params.static_friction = 534.0
plane_params.dynamic_friction = 5675.0
gym.add_ground(sim, plane_params)

# set up the env grid
num_envs = 1
spacing = 1.5
env_lower = gymapi.Vec3(-spacing, 0.0, -spacing)
env_upper = gymapi.Vec3(spacing, spacing, spacing)

# object
asset_options = gymapi.AssetOptions()
asset_options.density = 100.0
asset_capsule = gym.create_capsule(sim, 0.2, 0.2, asset_options)

env = gym.create_env(sim, env_lower, env_upper, 1)
pose = gymapi.Transform()
pose.p = gymapi.Vec3(0.0, 0.5, 0.0)
pose.r = gymapi.Quat(0, 0, 0, 1)
pose.p = gymapi.Vec3(0.0, 2.0, 0.0)
capsule_handle = gym.create_actor(env, asset_capsule, pose, name="object")

# set restitution for capsule actor
shape_props = gym.get_actor_rigid_shape_properties(env, capsule_handle)
default_props = gym.get_actor_rigid_shape_properties(env, capsule_handle)
shape_props[0].compliance = 999.0
shape_props[0].contact_offset = 999.0
shape_props[0].filter = 999
shape_props[0].friction = 999.0
shape_props[0].rest_offset = 999.0
shape_props[0].restitution = 999.0
shape_props[0].rolling_friction = 999.0
shape_props[0].thickness = 999.0
shape_props[0].torsion_friction = 999.0
gym.set_actor_rigid_shape_properties(env, capsule_handle, shape_props)

out_props = gym.get_actor_rigid_shape_properties(env, capsule_handle)

porperties_list = [
    "compliance",
    "contact_offset",
    "filter",
    "friction",
    "rest_offset",
    "restitution",
    "rolling_friction",
    "thickness",
    "torsion_friction",
]

print()
print()
print(f"{'Properties':18} {'Default':25} {'Input':8} {'Output':8}")
print(f"{'-' * 73}")
for prop in porperties_list:
    print(
        f"{prop:18} {default_props[0].__getattribute__(prop):25} {shape_props[0].__getattribute__(prop):8} {out_props[0].__getattribute__(prop):8}"
    )

# look at the first env
cam_pos = gymapi.Vec3(6, 4.5, 3)
cam_target = gymapi.Vec3(-0.5, 0.5, 0)
gym.viewer_camera_look_at(viewer, None, cam_pos, cam_target)

while not gym.query_viewer_has_closed(viewer):

    # step the physics
    gym.simulate(sim)
    gym.fetch_results(sim, True)

    # update the viewer
    gym.step_graphics(sim)
    gym.draw_viewer(viewer, sim, True)

    # Wait for dt to elapse in real time.
    # This synchronizes the physics simulation with the rendering rate.
    gym.sync_frame_time(sim)

print()
print()
print(f"{'Properties':18} {'Default':25} {'Input':8} {'Output':8}")
print(f"{'-' * 73}")
for prop in porperties_list:
    print(
        f"{prop:18} {default_props[0].__getattribute__(prop):25} {shape_props[0].__getattribute__(prop):8} {out_props[0].__getattribute__(prop):8}"
    )

gym.destroy_viewer(viewer)
gym.destroy_sim(sim)
```

Output:
```
Not connected to PVD
Physics Engine: PhysX
Physics Device: cpu
GPU Pipeline: disabled

Properties         Default                   Input    Output  
-------------------------------------------------------------------------
compliance                               0.0    999.0      0.0
contact_offset          0.019999999552965164    999.0    999.0
filter                                     0      999      999
friction                                 1.0    999.0    999.0
rest_offset            0.0010000000474974513    999.0    999.0
restitution                              0.0    999.0    999.0
rolling_friction                         0.0    999.0      0.0
thickness                                0.0    999.0      0.0
torsion_friction                         0.0    999.0      0.0
```

jagdeepsb, 2024-08-15:
Was there any resolution here? Experiencing the same issue

johannespitz, 2024-08-16:
As far as I know they are not planning to fix the problem. If you need to set some of these properties you'll have to re-implement everything in Isaac Sim (Omniverse).

closed 2024-08-16 by johannespitz