isaac-sim/IsaacGymEnvs #112: Exporting trained policy for sim2real

carebare47, 2023-02-10:
Hi,

I'm having difficulties in trying to take the trained policy out of Isaac/rl_games to use on a real robot. We've seen a project where someone builds ROS functionality into an isaac environment, but due to performance/deployment complexities this doesn't really work for us.

We've tried saving the model by accessing the runner [here](https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/blob/dee7c56765e14f6f4344c4d2e91d7a9eb3bfa619/isaacgymenvs/train.py#L156) with `torch.save(runner.model, '/path')` (after making a slight modification to `torch_runner.py` which returns the `player` from [here ](https://github.com/Denys88/rl_games/blob/fff05b72f78557f7acc932c25959404182c6c6b7/rl_games/torch_runner.py#L109) instead of calling `.run()` on it)

This didn't work, as far as I can tell due to anonymous references created by lambda functions in the model/network factories. We could start replacing these, but when you get down to the actual network itself it is constructed from lambdas (e.g, [here](https://github.com/Denys88/rl_games/blob/fff05b72f78557f7acc932c25959404182c6c6b7/rl_games/algos_torch/network_builder.py#L37-L57)), and rebuilding that in a way that is serializable seems excessive. 

Another approach could be to modify rl_games to collect an example input tensor(s) (e.g. [here](https://github.com/Denys88/rl_games/blob/fff05b72f78557f7acc932c25959404182c6c6b7/rl_games/algos_torch/a2c_continuous.py#L68-L70)), perform a torch script trace with the example input data, and save the full model (weights and architecture) in that program. This solution isn't great for us as we'd either have to maintain a fork of rl games or use patches to modify the rl_games code, just to let us save/export models. It would be great to be able to save the full model via the higher level interfaces.

A third solution could be to build rl_games into our real robot controllers, pass it the same config used to train the policy (to generate the same network architecture), and the load the checkpoint file saved by isaac/rl_games during training. Maybe this would be the most appropriate approach, although for some projects we are doing having to include this overhead isn't ideal.

Is there a nicer approach than these that I have missed? Is there a way to save the full model (architecture + state) from the higher level interface? Or more specifically, if you wanted to take this trained policy and deploy it outside of an isaac gym environment, how would you go about this?

Thanks,
Tom

eferreirafilho, 2024-03-18:
Have you managed to find a good way to achieve this?

Many thanks!

ronaldluc, 2024-03-19:
We've successfully deloyed models on real robots by embedding rl_games in
the ROS node inferencing the policy.

We had to also modify rl_games directly. It runs well on OrangePi5B cpu.

YuuChennABALONE, 2024-06-07:
How did you do it? Could you give me some advice?