isaac-sim/IsaacGymEnvs #117: GPU Pipeline: disabled, Not connected to PVD

dengyh16code, 2023-02-24:
I tried the examples “python joint_monkey.py” and got error. It told me that GPU Pipeline is disabled. It’s on Ubuntu 18.04.1 and RTX3090.
How can i enable my GPU pipeline?
Thankful for your help!

Error message as followings:

Importing module 'gym_37' (/home/robo/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_37.so)
Setting GYM_USD_PLUG_INFO_PATH to /home/robo/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
WARNING: Forcing CPU pipeline.
Not connected to PVD
+++ Using GPU PhysX
Physics Engine: PhysX
Physics Device: cuda:0
GPU Pipeline: disabled
Segmentation fault (core dumped)

The result for “nvidia-smi” is as followings:

Fri Feb 24 10:59:23 2023       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 525.89.02    Driver Version: 525.89.02    CUDA Version: 12.0     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|                               |                      |               MIG M. |
|===============================+======================+======================|
|   0  NVIDIA GeForce ...  Off  | 00000000:65:00.0  On |                  N/A |
| 32%   35C    P8    29W / 350W |    258MiB / 24576MiB |     12%      Default |
|                               |                      |                  N/A |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                                  |
|  GPU   GI   CI        PID   Type   Process name                  GPU Memory |
|        ID   ID                                                   Usage      |
|=============================================================================|
|    0   N/A  N/A      1415      G   /usr/lib/xorg/Xorg                 20MiB |
|    0   N/A  N/A      2162      G   /usr/lib/xorg/Xorg                235MiB |
+-----------------------------------------------------------------------------+

And the cuda 11.6 is also available in pytorch:

Python 3.7.12 | packaged by conda-forge | (default, Oct 26 2021, 06:08:53) 
[GCC 9.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import torch
>>> print(torch.version.cuda)
11.6

I have no idea how to fix this problem.

APiaoG, 2023-04-24:
Hello, I have also encountered the same problem. Have you resolved this issue?

Geni-sis, 2023-07-25:
Facing the same issue... any solution yet?

kc-ustc, 2023-11-07:
have you solved this problem?

qureshinomaan, 2024-01-11:
Has anyone been able to solve this?

qureshinomaan, 2024-01-15:
I was able to solve this by installing nvcc 
```
sudo apt install nvidia-cuda-toolkit
```

lurenjia384, 2024-07-15:
I solve it by next steps.
1.sudo apt-get install cmake git gcc g++ mesa-* libwayland-dev libxrandr-dev
sudo apt-get install libvulkan1 mesa-vulkan-drivers vulkan-utils
2.cd conda env/lib:cd /usr/local/anaconda3/envs/ycc_rlgym/ilb
rm libffi.so.7
sudo ln -s /lib/x86_64-linux-gnu/libffi.so.7.1.0 libffi.so.7
3.all is ok!!!