isaac-sim/IsaacGymEnvs #124: May be a bug in setup.py

RayYoh, 2023-03-25:
A bug in `setup.py` when I pip this package. I'm not sure I'm a special case. I commented out that line, and it works.
```
reparing metadata (setup.py) ... error
  error: subprocess-exited-with-error
  
  × python setup.py egg_info did not run successfully.
  │ exit code: 1
  ╰─> [1 lines of output]
      error in isaacgymenvs setup command: 'python_requires' must be a string containing valid version specifiers; Invalid specifier: '>=3.6.*'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.
```

ccmCCMfk, 2023-08-26:
I have met the same problem with you,do you find a solution now?

ccmCCMfk, 2023-08-26:
You can just delete "python_requires=">=3.6","

RayYoh, 2023-08-29:
> You can just delete "python_requires=">=3.6","

Yes, I commented out it and it worked.

TAEM1N2, 2024-01-19:
maybe you should check your internet connection. I met an same error, and solved it