isaac-sim/IsaacGymEnvs #125: Funny Bug

GuoPingPan, 2023-03-26:


hankpan99, 2023-04-02:
Hi, I met the same bug, have you found out any solution, thanks!!!

clwmrndl92, 2023-06-03:
> 

If you use multiple actors, when using "set_dof_state_tensor_indexed", it works well if you use only the index of the actor with dof.

 For example..

```
        env_ids_int32 = env_ids.to(dtype=torch.int32)
        self.gym.set_actor_root_state_tensor_indexed(self.sim,
                                                     gymtorch.unwrap_tensor(self.root_state),
                                                     gymtorch.unwrap_tensor(env_ids_int32), len(env_ids_int32))

        human_indices = self.gym_indices["human"][env_ids].to(torch.int32)
        self.gym.set_dof_state_tensor_indexed(self.sim,
                                              gymtorch.unwrap_tensor(self.dof_state),
                                              gymtorch.unwrap_tensor(human_indices), len(human_indices))
```

See "trifinger" example env code