isaac-sim/IsaacGymEnvs #126: vulkan: No DRI3 support detected - required for presentation

StoneT2000, 2023-04-06:
I'm getting this message when I run the example code in the readme that generates videos. It seems in headless=False mode it can't open the viewer and the env.render function returns None.

Any idea how to fix this?

I am on ubuntu 22.04, RTX 3090Ti

gavrielstate, 2023-04-14:
You should ensure that you have working vulkan on your system. Try `vulkaninfo` and `vkcube` to make sure they work. You may need to blacklist drivers for other GPUs to avoid an integrated GPU getting picked up as vulkan device 0.

kinalmehta, 2023-11-14:
Getting the same error on fedora 39 rtx 3060. Anyone solved this?

tomtang502, 2024-05-20:
Same error with Ubuntu with 4090, and seem like this is a persistent issue?

paLeziart, 2024-06-03:
@StoneT2000 @kinalmehta @tomtang502 I had to dig a bit everywhere to find the cause of this issue and I finally found it: Vulkan is not selecting the right device by default and is trying to render on the CPU (which likely does not support DRI3) instead of the GPU. I found the solution on a forum for Arch Linux but the solution will likely work for all kinds of Linux distros, like for me on a derivative of Ubuntu 22.04.

* Install 'vulcaninfo': `sudo apt install vulkan-tools`

* Identify your GPU device: `MESA_VK_DEVICE_SELECT=list vulkaninfo`
* You will get something like that, with your GPU and your iGPU.
```
GPU 0: xxxxx:0 "llvmpipe [..]"
GPU 1: abcd:efgh "NVIDIA RTX [..]"
GPU 2: ijkl:mnop "Intel(R) Graphics [..]"
```
* Force the Vulkan device when running Isaac: `MESA_VK_DEVICE_SELECT=abcd:efgh python3 train.py task=[...]`

* I guess you can also set the env variable at the end of your `~/.bashrc` if you want to always force it: `export MESA_VK_DEVICE_SELECT="abcd:efgh"`

**Sources of the solution:**
Archi Linux wiki: [https://wiki.archlinux.org/title/Vulkan#Switching_between_devices](https://wiki.archlinux.org/title/Vulkan#Switching_between_devices)
Mesa3D documentation: [https://docs.mesa3d.org/envvars.html#vulkan-mesa-device-select-layer-environment-variables](https://docs.mesa3d.org/envvars.html#vulkan-mesa-device-select-layer-environment-variables)

paLeziart, 2024-06-03:
A funny thing to note is that running `vkcube` outputs
```
$ vkcube
Selected GPU 2: NVIDIA RTX [....] GPU, type: 2
```
even though the Nvidia GPU was listed as GPU 1 with `MESA_VK_DEVICE_SELECT=list vulkaninfo` (GPU 2 being the integrated GPU). This might be linked to the issue.

closed 2024-06-03 by StoneT2000