isaac-sim/IsaacGymEnvs #129: Trying to run on cuda:1 crashes

Robokan, 2023-04-14:
**I have 2 GPU's and I want to only train on the second one so I ran:

python train.py task=Cartpole rl_device='cuda:1' sim_device='cuda:1'

it crashes saying I am still running something on cuda:0. Any ideas how to fix this?

here is the full stack trace:**

(rlenv) bizon@dl:~/eric/IsaacGymEnvs-main/isaacgymenvs$ python train.py task=Cartpole rl_device='cuda:1' sim_device='cuda:1'
Importing module 'gym_38' (/home/bizon/anaconda3/envs/rlenv/lib/python3.8/site-packages/isaacgym/_bindings/linux-x86_64/gym_38.so)
Setting GYM_USD_PLUG_INFO_PATH to /home/bizon/anaconda3/envs/rlenv/lib/python3.8/site-packages/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
train.py:49: UserWarning: 
The version_base parameter is not specified.
Please specify a compatability version level, or None.
Will assume defaults for version 1.1
  @hydra.main(config_name="config", config_path="./cfg")
/home/bizon/anaconda3/envs/rlenv/lib/python3.8/site-packages/hydra/_internal/defaults_list.py:251: UserWarning: In 'config': Defaults list is missing `_self_`. See https://hydra.cc/docs/1.2/upgrades/1.0_to_1.1/default_composition_order for more information
  warnings.warn(msg, UserWarning)
/home/bizon/anaconda3/envs/rlenv/lib/python3.8/site-packages/hydra/_internal/defaults_list.py:415: UserWarning: In config: Invalid overriding of hydra/job_logging:
Default list overrides requires 'override' keyword.
See https://hydra.cc/docs/1.2/upgrades/1.0_to_1.1/defaults_list_override for more information.

  deprecation_warning(msg)
/home/bizon/anaconda3/envs/rlenv/lib/python3.8/site-packages/hydra/_internal/hydra.py:119: UserWarning: Future Hydra versions will no longer change working directory at job runtime by default.
See https://hydra.cc/docs/1.2/upgrades/1.1_to_1.2/changes_to_job_working_dir/ for more information.
  ret = run_job(
PyTorch version 1.13.1
Device count 2
/home/bizon/anaconda3/envs/rlenv/lib/python3.8/site-packages/isaacgym/_bindings/src/gymtorch
Using /home/bizon/.cache/torch_extensions/py38_cu117 as PyTorch extensions root...
Emitting ninja build file /home/bizon/.cache/torch_extensions/py38_cu117/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
Loading extension module gymtorch...
/home/bizon/anaconda3/envs/rlenv/lib/python3.8/site-packages/isaacgym/torch_utils.py:135: DeprecationWarning: `np.float` is a deprecated alias for the builtin `float`. To silence this warning, use `float` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.float64` here.
Deprecated in NumPy 1.20; for more details and guidance: https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
  def get_axis_params(value, axis_idx, x_value=0., dtype=np.float, n_dims=3):
2023-04-14 09:06:54,989 - INFO - logger - logger initialized
<unknown>:3: DeprecationWarning: invalid escape sequence \*
Error: FBX library failed to load - importing FBX data will not succeed. Message: No module named 'fbx'
FBX tools must be installed from https://help.autodesk.com/view/FBX/2020/ENU/?guid=FBX_Developer_Help_scripting_with_python_fbx_installing_python_fbx_html
/home/bizon/anaconda3/envs/rlenv/lib/python3.8/site-packages/torch/utils/tensorboard/__init__.py:4: DeprecationWarning: distutils Version classes are deprecated. Use packaging.version instead.
  if not hasattr(tensorboard, "__version__") or LooseVersion(
/home/bizon/anaconda3/envs/rlenv/lib/python3.8/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:568: DeprecationWarning: `np.object` is a deprecated alias for the builtin `object`. To silence this warning, use `object` by itself. Doing this will not modify any behavior and is safe. 
Deprecated in NumPy 1.20; for more details and guidance: https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
  (np.object, string),
/home/bizon/anaconda3/envs/rlenv/lib/python3.8/site-packages/tensorboard/compat/tensorflow_stub/dtypes.py:569: DeprecationWarning: `np.bool` is a deprecated alias for the builtin `bool`. To silence this warning, use `bool` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.bool_` here.
Deprecated in NumPy 1.20; for more details and guidance: https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
  (np.bool, bool),
/home/bizon/anaconda3/envs/rlenv/lib/python3.8/site-packages/tensorboard/util/tensor_util.py:100: DeprecationWarning: `np.object` is a deprecated alias for the builtin `object`. To silence this warning, use `object` by itself. Doing this will not modify any behavior and is safe. 
Deprecated in NumPy 1.20; for more details and guidance: https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
  np.object: SlowAppendObjectArrayToTensorProto,
/home/bizon/anaconda3/envs/rlenv/lib/python3.8/site-packages/tensorboard/util/tensor_util.py:101: DeprecationWarning: `np.bool` is a deprecated alias for the builtin `bool`. To silence this warning, use `bool` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.bool_` here.
Deprecated in NumPy 1.20; for more details and guidance: https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
  np.bool: SlowAppendBoolArrayToTensorProto,
task: 
    name: Cartpole
    physics_engine: physx
    env: 
        numEnvs: 512
        envSpacing: 4.0
        resetDist: 3.0
        maxEffort: 400.0
        clipObservations: 5.0
        clipActions: 1.0
        asset: 
            assetRoot: ../../assets
            assetFileName: urdf/cartpole.urdf
        enableCameraSensors: False
[... 129 lines cut ...]
  File "train.py", line 161, in launch_rlg_hydra
    runner.run({
  File "/home/bizon/anaconda3/envs/rlenv/lib/python3.8/site-packages/rl_games/torch_runner.py", line 120, in run
    self.run_train(args)
  File "/home/bizon/anaconda3/envs/rlenv/lib/python3.8/site-packages/rl_games/torch_runner.py", line 101, in run_train
    agent.train()
  File "/home/bizon/anaconda3/envs/rlenv/lib/python3.8/site-packages/rl_games/common/a2c_common.py", line 1173, in train
    step_time, play_time, update_time, sum_time, a_losses, c_losses, b_losses, entropies, kls, last_lr, lr_mul = self.train_epoch()
  File "/home/bizon/anaconda3/envs/rlenv/lib/python3.8/site-packages/rl_games/common/a2c_common.py", line 1037, in train_epoch
    batch_dict = self.play_steps()
  File "/home/bizon/anaconda3/envs/rlenv/lib/python3.8/site-packages/rl_games/common/a2c_common.py", line 626, in play_steps
    res_dict = self.get_action_values(self.obs)
  File "/home/bizon/anaconda3/envs/rlenv/lib/python3.8/site-packages/rl_games/common/a2c_common.py", line 348, in get_action_values
    res_dict = self.model(input_dict)
  File "/home/bizon/anaconda3/envs/rlenv/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1194, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/bizon/anaconda3/envs/rlenv/lib/python3.8/site-packages/rl_games/algos_torch/models.py", line 246, in forward
    input_dict['obs'] = self.norm_obs(input_dict['obs'])
  File "/home/bizon/anaconda3/envs/rlenv/lib/python3.8/site-packages/rl_games/algos_torch/models.py", line 49, in norm_obs
    return self.running_mean_std(observation) if self.normalize_input else observation
  File "/home/bizon/anaconda3/envs/rlenv/lib/python3.8/site-packages/torch/nn/modules/module.py", line 1194, in _call_impl
    return forward_call(*input, **kwargs)
  File "/home/bizon/anaconda3/envs/rlenv/lib/python3.8/site-packages/rl_games/algos_torch/running_mean_std.py", line 79, in forward
    y = (input - current_mean.float()) / torch.sqrt(current_var.float() + self.epsilon)
RuntimeError: Expected all tensors to be on the same device, but found at least two devices, cuda:1 and cuda:0!

HuSha3k, 2023-04-25:
Hi, I encountered the same issue, and according to #109, it is because `rl_device='cuda:1'` doesn't work correctly. 

you can either follow their solution or simply add `CUDA_VISIBLE_DEVICES=[gpu_ids]` infront of your training command.

Robokan, 2023-04-30:
I just tried

CUDA_VISIBLE_DEVICES=1, python train.py task=Cartpole 
CUDA_VISIBLE_DEVICES=1, python train.py task=Cartpole rl_device='cuda:1' sim_device='cuda:1'
CUDA_VISIBLE_DEVICES=[1], python train.py task=Cartpole rl_device='cuda:1' sim_device='cuda:1'
CUDA_VISIBLE_DEVICES=[1], python train.py task=Cartpole

None of these work. It still crashes. I tried just using export as well. Were you able to get it to work?

Robokan, 2023-05-12:
Great that works. Thanks for the clarification.

closed 2023-05-12 by Robokan