isaac-sim/IsaacGymEnvs #131: BUG CRASH when recording video (python train.py task=HumanoidAMP capture_video=True) --> render() got an unexpected keyword argument 'mode'

generush, 2023-04-21:
Hello,

I've discovered IsaacGymEnvs can capture video (capture_video=True) for static viewers, but not dynamic ones.

IsaacGymEnvs immediately crashes when I run:

```bash
python train.py task=HumanoidAMP capture_video=True
```

Here is the error it produces:

```bash
Error executing job with overrides: ['task=HumanoidAMP', 'capture_video=True']
Traceback (most recent call last):
  File "train.py", line 190, in launch_rlg_hydra
    runner.run({
  File "/home/gene/code/NEURO/neuro-rl-sandbox/rl_games/rl_games/torch_runner.py", line 121, in run
    self.run_train(args)
  File "/home/gene/code/NEURO/neuro-rl-sandbox/rl_games/rl_games/torch_runner.py", line 102, in run_train
    agent.train()
  File "/home/gene/code/NEURO/neuro-rl-sandbox/IsaacGymEnvs/isaacgymenvs/learning/common_agent.py", line 120, in train
    self.obs = self.env_reset()
  File "/home/gene/code/NEURO/neuro-rl-sandbox/rl_games/rl_games/common/a2c_common.py", line 507, in env_reset
    obs = self.vec_env.reset()
  File "/home/gene/code/NEURO/neuro-rl-sandbox/IsaacGymEnvs/isaacgymenvs/utils/rlgames_utils.py", line 249, in reset
    return self.env.reset()
  File "/home/gene/anaconda3/envs/isaacgym/lib/python3.8/site-packages/gym/wrappers/record_video.py", line 58, in reset
    self.start_video_recorder()
  File "/home/gene/anaconda3/envs/isaacgym/lib/python3.8/site-packages/gym/wrappers/record_video.py", line 75, in start_video_recorder
    self.video_recorder.capture_frame()
  File "/home/gene/anaconda3/envs/isaacgym/lib/python3.8/site-packages/gym/wrappers/monitoring/video_recorder.py", line 157, in capture_frame
    frame = self.env.render(mode=render_mode)
TypeError: render() got an unexpected keyword argument 'mode'
```

I found this error, because I attempted to duplicate code from [HumanoidAMP](https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/blob/3d2d5ed7ce71401db3063d8c339f4a800ce8709f/isaacgymenvs/tasks/amp/humanoid_amp_base.py#L392) to AnymalTerrain, but it also crashes and produces the same error.

Searching online, it seems people have faced various issues relating to gym versioning/compatibility, but not this specific issue. I haven't been able to figure out a solution after debugging for a day or two.

I would appreciate it if someone could take a look and hopefully spot a quick solution! Thank you very much for your help : )

generush, 2023-04-21:
I did not mean to close this issue, as it is still unresolved. However, I wanted to include the package version in my environment:

```vim
# packages in environment at /home/gene/anaconda3/envs/isaacgym:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                        main  
_openmp_mutex             5.1                       1_gnu  
absl-py                   1.2.0                    pypi_0    pypi
aiosignal                 1.2.0                    pypi_0    pypi
altair                    4.2.2                    pypi_0    pypi
antlr4-python3-runtime    4.9.3                    pypi_0    pypi
anyio                     3.6.2                    pypi_0    pypi
argon2-cffi               21.3.0                   pypi_0    pypi
argon2-cffi-bindings      21.2.0                   pypi_0    pypi
arrow                     1.2.3                    pypi_0    pypi
astroid                   2.15.3                   pypi_0    pypi
asttokens                 2.0.5              pyhd3eb1b0_0  
attrs                     22.1.0                   pypi_0    pypi
autopep8                  1.7.0                    pypi_0    pypi
backcall                  0.2.0              pyhd3eb1b0_0  
backports-zoneinfo        0.2.1                    pypi_0    pypi
beautifulsoup4            4.11.2                   pypi_0    pypi
bleach                    6.0.0                    pypi_0    pypi
blinker                   1.5                      pypi_0    pypi
bokeh                     2.4.3                    pypi_0    pypi
boto3                     1.26.63                  pypi_0    pypi
botocore                  1.29.63                  pypi_0    pypi
ca-certificates           2023.01.10           h06a4308_0  
cachetools                5.2.0                    pypi_0    pypi
certifi                   2022.12.7        py38h06a4308_0  
cffi                      1.15.1                   pypi_0    pypi
cftime                    1.6.2                    pypi_0    pypi
charset-normalizer        2.1.1                    pypi_0    pypi
click                     8.0.4                    pypi_0    pypi
cloudpickle               2.2.0                    pypi_0    pypi
codecov                   2.1.12                   pypi_0    pypi
colorcet                  3.0.1                    pypi_0    pypi
coloredlogs               15.0.1                   pypi_0    pypi
comm                      0.1.2                    pypi_0    pypi
comparative-legged-rl     0.1.0                     dev_0    <develop>
contourpy                 1.0.5                    pypi_0    pypi
coverage                  7.1.0                    pypi_0    pypi
cycler                    0.11.0                   pypi_0    pypi
dash                      2.8.1                    pypi_0    pypi
dash-bootstrap-components 1.4.0                    pypi_0    pypi
dash-core-components      2.0.0                    pypi_0    pypi
dash-html-components      2.0.0                    pypi_0    pypi
dash-table                5.0.0                    pypi_0    pypi
dask                      2023.1.1                 pypi_0    pypi
datashader                0.14.4                   pypi_0    pypi
datashape                 0.5.2                    pypi_0    pypi
debugpy                   1.6.6                    pypi_0    pypi
decorator                 4.4.2                    pypi_0    pypi
deepdiff                  6.2.3                    pypi_0    pypi
deeplake                  3.2.4                    pypi_0    pypi
defusedxml                0.7.1                    pypi_0    pypi
dill                      0.3.6                    pypi_0    pypi
distlib                   0.3.6                    pypi_0    pypi
docker-pycreds            0.4.0                    pypi_0    pypi
einops                    0.6.0                    pypi_0    pypi
entrypoints               0.4                      pypi_0    pypi
exceptiongroup            1.1.0                    pypi_0    pypi
executing                 0.8.3              pyhd3eb1b0_0  
fastjsonschema            2.16.2                   pypi_0    pypi
ffmpeg                    1.4                      pypi_0    pypi
filelock                  3.8.0                    pypi_0    pypi
flake8                    6.0.0                    pypi_0    pypi
flask                     2.2.2                    pypi_0    pypi
flatbuffers               22.9.24                  pypi_0    pypi
[... 207 lines cut ...]
tqdm                      4.64.1                   pypi_0    pypi
traitlets                 5.9.0                    pypi_0    pypi
typing-extensions         4.4.0                    pypi_0    pypi
tzdata                    2022.7                   pypi_0    pypi
tzlocal                   4.2                      pypi_0    pypi
uri-template              1.2.0                    pypi_0    pypi
urllib3                   1.26.12                  pypi_0    pypi
validators                0.20.0                   pypi_0    pypi
virtualenv                20.16.5                  pypi_0    pypi
wandb                     0.12.11                  pypi_0    pypi
watchdog                  2.2.1                    pypi_0    pypi
wcwidth                   0.2.5              pyhd3eb1b0_0  
webcolors                 1.12                     pypi_0    pypi
webencodings              0.5.1                    pypi_0    pypi
websocket-client          1.5.1                    pypi_0    pypi
[... cut ...]
widgetsnbextension        4.0.5                    pypi_0    pypi
wrapt                     1.15.0                   pypi_0    pypi
xarray                    2023.1.0                 pypi_0    pypi
xz                        5.2.6                h5eee18b_0  
yaspin                    2.2.0                    pypi_0    pypi
zict                      2.2.0                    pypi_0    pypi
zipp                      3.9.0                    pypi_0    pypi
zlib                      1.2.12               h5eee18b_3

zzzark, 2023-11-29:
@generush Hi. I have encountered the same issue. After lots of debugging I find that it is not a bug in rl_games or gym or any other packages, but indeed in humanoid_amp_base.py.

So to fix this issue just simply change the following lines in humanoid_amp_base.py in line 392 from:

```python
def render(self):
    if self.viewer and self.camera_follow:
        self._update_camera()

    super().render()
    return
```

to 

```python
def render(self, mode="rgb_array"):
    if self.viewer and self.camera_follow:
        self._update_camera()

    return super().render(mode)
```

Hope this help.