isaac-sim/IsaacGymEnvs #135: SDF collision mesh rendering bug

lorenmt, 2023-05-02:
Just found a bug in collision mesh rendering from the nut and bolt SDF example.

The collision rendering seems to be sensitive to the order of loading assets.

The example file `franka_nut_bolt_ik_osc.py` loads the bolt and nut assets first and then the Franka asset. If we move the Franka asset loading code before the bolt and nut, the collision mesh rendering will be incorrect. But only the rendering is incorrect, the physics and the collision checking do not seem to be affected.

And I have confirmed that it's only happening in SDF meshes. The standard asset loading without SDF or just using convex decomposition would not be affected. 

The examples of correct and incorrect mesh rendering are provided below.

Also, adding `sdf` tags in the Franka urdf will also give the wrong rendering.

[image]
[image]

kirtan-raise, 2023-11-01:
Hi @lorenmt 
I am observing something similar in rendered collision meshes. My example is based on factory_task_nut_bolt_place.py and I am using Ur10e arm instead of Franka. I am using nut_bolt_m16_tight and nut_bolt_m16_loose for the task. 

Here, is the picture of an actual simulation. 
[image]

And here is the same simulation with the collision meshes turned on. So, mesh that gets loaded at the end has the biggest collision mesh rendering. So, turns out it is not specific to tight or loose nut and bolt assets. 

[image]

@lorenmt I was wondering how did you made sure that it is not affecting the physics and collision checking. Thanks.

lorenmt, 2023-11-01:
A quick stupid way is to change the order when you load the assets.