isaac-sim/IsaacGymEnvs #140: Error: FBX library failed to load - importing FBX data will not succeed. Message: No module named 'fbx'

yunlong-wang-cn, 2023-06-14:
When I try to run the following code:

	import gym
	import isaacgym
	import isaacgymenvs
	import torch

	envs = isaacgymenvs.make(
		seed=0,
		task="Ant",
		num_envs=20,
		sim_device="cuda:0",
		rl_device="cuda:0",
		graphics_device_id=0,
		headless=False,
		multi_gpu=False,
		virtual_screen_capture=True,
		force_render=False,
	)
	envs.is_vector_env = True
	envs = gym.wrappers.RecordVideo(
		envs,
		"./videos",
		step_trigger=lambda step: step % 10000 == 0, # record the videos every 10000 steps
		video_length=100  # for each video record up to 100 steps
	)
	envs.reset()
	print("the image of Isaac Gym viewer is an array of shape", envs.render(mode="rgb_array").shape)
	for _ in range(100):
		envs.step(
			torch.rand((20,)+envs.action_space.shape, device="cuda:0")
		)
Then the generated video file is empty and report following error:

    Error: FBX library failed to load - importing FBX data will not succeed. Message: No module named 'fbx'
    FBX tools must be installed from https://help.autodesk.com/view/FBX/2020/ENU/? 
    guid=FBX_Developer_Help_scripting_with_python_fbx_installing_python_fbx_html

Then by the given link, it told me to try to install the[ fbx python binding](https://www.autodesk.com/content/dam/autodesk/www/adn/fbx/2020-3-4/fbx202034_fbxpythonbindings_linux.tar.gz), there is no wheel file, which should be there. And everything seems get stuck.

dibbla, 2023-06-15:
I also have trouble in recording the video, but is it really related to FBX? 👀

maths4513, 2023-06-20:
I got the same issue - 'fbx error'. But in the end it seems has nothing to do with fbx python binding at all.

------------------------
Ubuntu: 20.04
Python: 3.8
Nvidia Driver: 450.
------------------------
**Erros**:
1.fbx moudle import error.
2.GPU Pipe Line disabled.

**How the problem is gone**:
1.Update nvidia driver from 450 to 510.
2.Reboot
-----------------------

gavrielstate, 2023-06-20:
The FBX Importer error message is due to the fact that we import the skeleton code as part of the task initialization, but it's only used for the AMP examples:
 
https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/blob/main/docs/rl_examples.md#amp-adversarial-motion-priors-humanoidamppy

You can ignore the FBX error for other environments if you don't have it installed.

yunlong-wang-cn, 2023-06-20:
As other people said, perhaps it was caused by another issue. Thanks. I will check my code.

yqi19, 2023-11-20:
Hi, have you solved this? I am running into the same issue!

yxt181110417, 2024-09-26:
error1 :  FBX library failed to load - importing FBX data will not succeed.
error2:  other problems

i solve the error2,but error1 is still exist.  And the isaacgymenvs example is can still works.   i run the task == Trifinger and task == Ant . 
So, maybe we can now ignore the FBX error , and continue learning