isaac-sim/IsaacGymEnvs #145: Fail to create 2 environments

glx15534565855, 2023-07-02:
When I run the following code：

```
import isaacgym
import isaacgymenvs

anymal_env = isaacgymenvs.make(seed=0, task="Anymal", num_envs=1, sim_device="cuda:0", rl_device="cuda:0", graphics_device_id=0,headless=True,multi_gpu=False)
ant_env = isaacgymenvs.make(seed=0, task="Ant", num_envs=1, sim_device="cuda:0", rl_device="cuda:0", graphics_device_id=0, headless=True,multi_gpu=False)

```

it reports the error:

```
The version_base parameter is not specified.
Please specify a compatability version level, or None.
Will assume defaults for version 1.1
  with initialize(config_path="./cfg"):
num envs 1 env spacing 5
/home/guolingxiao/anaconda3/IsaacGymEnvs/isaacgymenvs/tasks/ant.py:151: DeprecationWarning: an integer is required (got type isaacgym._bindings.linux-x86_64.gym_38.DofDriveMode).  Implicit conversion to integers using __int__ is deprecated, and may be removed in a future version of Python.
  asset_options.default_dof_drive_mode = gymapi.DOF_MODE_NONE
[Error] [carb.gym.plugin] Function GymGetActorDofStates cannot be used with the GPU pipeline after simulation starts.  Please use the tensor API if possible.  See docs/programming/tensors.html for more info.

```

But when I run this:

```
import isaacgym
import isaacgymenvs

anymal_env = isaacgymenvs.make(seed=0, task="Anymal", num_envs=1, sim_device="cuda:0", rl_device="cuda:0", graphics_device_id=0,headless=True,multi_gpu=False)

```

it just runs successfully.

So I wonder if the isaacgym cannot create 2 envs at one run, or if there are some restrictions that I can cancel to make it through?

LyuJZ, 2023-11-29:
Same error!

glx15534565855, 2023-11-29:
It seems that  isaacgym can only make envs at once. If you want to create different robots at once, I suggest you to invoke " gym.create_actor()" to create multiple robots. Although isaacgym can only create env one time, it can add multiple actors to one environment. Refer to the isaacgym/docs/index.html in your isaacgym package for detailed usage. Hope that helps