isaac-sim/IsaacGymEnvs #157: Add to documentation: When to use IsaacGymEnvs and OmniIsaacGymEnvs

sambaPython24, 2023-07-31:
Hey, 
there is the ```IsaacGymEnvs``` environment and the ```OmniIsaacGymEnvs``` environment in the ```NVIDIA-Omniverse``` repo.
Could you write down explicitely how to choose between them?

DanielTakeshi, 2023-08-02:
Agree that this would probably be helpful.

eferreirafilho, 2024-03-18:
From Isaac Sim Discord channel:

> 
> Taking the easiest one first, there is no "standalone Isaac Sim."  For as long as Isaac Sim has been publicly available, it's been built on Omniverse.
> The same can't be said about Isaac Gym, which was initially available as a standalone "Preview" release.  Most of the functionality of standalone Isaac Gym has been moved to the Gym extension within (Omniverse) Isaac Sim, which is what will be supported and developed going forward.
> 
> Orbit is a set of interfaces and environments that build on top of Isaac Sim (including the Gym extension within Isaac Sim).  It's meant to simplify certain common workflows and provide a much more extensive set of sample environments than can be found in Isaac Sim proper.  We also foresee it evolving with contributions from the community.  Orbit vs. Isaac Sim isn't really an either/or decision, since the former builds on the latter.  My suggestion would be to peruse the environments in Orbit and consider using it if your use case is similar.
> 
> For completeness, I should also mention Omniverse Isaac Gym Envs (OIGE), which is a set of environments more specifically focused on reinforcement learning.  They leverage the Isaac Sim core and gym APIs but not Orbit.  There's definitely some overlap between OIGE and Orbit, and we're hoping to harmonize them in the future.
> 
> I am a robotics researcher in academia & also lead the Orbit work at Nvidia. 
> We are cognizant of the technical problems faced by the community in robot learning and are working towards unified solutions. 
> 
> Omniverse Isaac-sim (primary parent product)- enterprise grade simulation in robotics
> - Isaac-sim (standalone): this is not a thing. apologies if this was confusing.
> 
> - IsaacGym - initial interface built to showcase physX with tensorAPI. Hence this was always a preview version. Importantly Isaac-Gym is not under development any more. This functionality is now in the IsaacSim. All future development and features will be in IsaacSim.
> 
> - Omniverse Isaac-gym (OIGE)- Minimal samples to showcase that IG style tensorAPI is now available in IsaacSim.
> 
> - Isaac Orbit- Batteries included framework to reduce barrier to entry. It serves a dual purpose: 
>    (1) simplified interface for env design and support for many robotics workflows - RL, Motion planning, teleoperation, imitation learning/behavior cloning, and real robot operation. This unification is the USP of orbit as compared to other interfaces.
>    (2) Front end framework for prebuilt environments as benchmark examples. Examples similar to OIGE have already been tested in Orbit. 
> 
> We note that many folks in the community are users of IsaacSim, creating new environments rather than physics solvers. 
> Hence, we hope to provide Orbit to warmstart it with prebuilt assets, and environment examples as benchmarks.
> 
> Importantly Orbit is designed such that it can accept community contributions with open licensing. 
> We hope that Orbit will be the environment zoo for IsaacSim with contributions from the community as well as internal development. 
> 
> Orbit also has modified env interfaces. However, Orbit is open source, and users can modify & suggest change to these interfaces, as needed. Further additional support to new features such as new RL/Motion Planning libraries can also be added through a PR.