isaac-sim/IsaacGymEnvs #161: What is the exact formula for the set_dof_position_target command?

sizsJEon, 2023-08-28:
currently facing difficulties while trying to transfer the results obtained using position control in Isaac Sim to a different environment
And I need the formula for the value ultimately calculated in the set_dof_position_target fuction provided by Isaac.

Some posts suggest that it is a general PD control, but the results I calculated are different from this, and causing confusion.

https://forums.developer.nvidia.com/t/torque-control-implementation/182378/8?u=sizs2002
https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/blob/665c32170d84b4be66722eea405a1e08b6e7f761/isaacgymenvs/tasks/trifinger.py#L1021C1-L1022C85

The detailed internal source code of the API is not publicly available, which is why I am posting this topic.
I am looking forward to receiving many responses.

(I posted this issue in nvidia forum but got no response from it...)

gavrielstate, 2023-08-31:
These parameters are passed down to the PhysX engine through the GPU APIs (https://nvidia-omniverse.github.io/PhysX/physx/5.2.1/docs/Articulations.html#data-oriented-direct-gpu-api), but you probably want to look more closely at the articulation documentation here: https://nvidia-omniverse.github.io/PhysX/physx/5.2.1/docs/Articulations.html#articulation-joint-drives

You will find the CPU solver source code in here:
https://github.com/NVIDIA-Omniverse/PhysX/tree/release/104.2/physx/source/lowleveldynamics/src

sizsJEon, 2023-09-05:
thanks for the information! I'll check it out.

closed 2023-09-05 by sizsJEon