isaac-sim/IsaacGymEnvs #179: A potential error in humanoid_amp

Charrrrrlie, 2023-10-31:
Hello,

When calling [_reset_ref_state_init function](https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/blob/aeed298638a1f7b5421b38f5f3cc2d1079b6d9c3/isaacgymenvs/tasks/humanoid_amp.py#L182C9-L182C30), the `rigid_body_state` in [_set_env_state](https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/blob/aeed298638a1f7b5421b38f5f3cc2d1079b6d9c3/isaacgymenvs/tasks/humanoid_amp.py#L197)  will not be updated. And the base observations in [humanoid_amp_base](https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/blob/aeed298638a1f7b5421b38f5f3cc2d1079b6d9c3/isaacgymenvs/tasks/amp/humanoid_amp_base.py#L149) will be calculated with a default/old rigid body state. 

That may cause a potential error when we leverage the base observations in the simulation. 
e.g. call self._rigid_body_state in pre_physics_step.
  (The current AMP algorithm uses the observations in `humanoid_amp.py`, thus it won't effect on the results)

Moreover, I cannot add functions like `set_rigid_body_state_tensor` in _set_env_state since the simulation is under Physx. So…In my case, I have to manually change the value of self._rigid_body_state and force it not to refresh in [reset_actors](https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/blob/aeed298638a1f7b5421b38f5f3cc2d1079b6d9c3/isaacgymenvs/tasks/amp/humanoid_amp_base.py#L146C9-L146C18). 

Is there any solution if it is indeed a potential error?

Thanks for any help!

heyuanYao-pku, 2024-01-22:
Same question, any solution now?

Charrrrrlie, 2024-01-23:
> Same question, any solution now?

Not yet 
:-(

And there are so many differences compared with the implementation of ASE.  e.g. [this problem](https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/pull/198) you may also need to pay attention to it.