isaac-sim/IsaacGymEnvs #181: Physical Dynamics Variation in Factory and IndustReal Environments

patricknaughton01, 2023-11-01:
Hello,

I've been using IsaacGym and in particular the IndustReal environments to learn peg insertion policies. I was hoping to extend these environments to simulate scenes that vary physical parameters (specifically the friction of the socket and plug, and the mass of the plug) between environment resets. Unfortunately, it seems that trying to adjust the friction of the socket (via `set_actor_rigid_shape_properties​`) in the `reset_idx`​ function causes the simulator to ignore collisions between the peg and socket. The problem seems to happen even if I simply read the sockets' properties and then set them back to the same values. 

[image]
(socket friction read and set back to same value).

Additionally, trying to read and reset the mass of the plug back to the same value seems to cause the simulator to miss a position update. Here's a video of a "normal" execution (without trying to write to the plug's mass):

https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/assets/28968518/56f60ad5-0711-4b21-acf8-979d193e131f

And after trying to read and reset the plugs mass:

https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/assets/28968518/5048611c-cf69-419f-a22d-7a87dc1c1eae

I was wondering if I'm maybe doing something wrong and I should be able to modify these parameters during `reset_idx`​ with the SDF geometries. I wasn't sure because I don't think the Factory or IndustReal project did this. Since the SDF representation seems to be relatively new, I was also wondering if there's something I could do on my end to integrate this functionality. 

Thanks for your help!

bingjietang718, 2023-11-07:
Hi Patrick,

Can you try add `self.simulate_and_refresh()` every time right after you modify physics parameters? This function will essentially simulate one step and apply the changes you made to the physical environment.

Best,
Bingjie

patricknaughton01, 2023-11-07:
Hi Bingjie,

Quick note is that I did try to call `self.simulate_and_refresh` after modifying the parameters, it fixed one of the issues but introduced a new one:

I did some more poking around to characterize the issue a bit more. For all of the below scenarios, I ran this command:
`python train.py task=IndustRealTaskPegsInsert task.rl.curriculum_height_bound=[0.005,0.01]`

with this `reset_idx` (keeping gravity off so I can see the peg fall into the socket):
<pre>
def reset_idx(self, env_ids):
      """Reset specified environments."""

      self._reset_franka()

      # Close gripper onto plug
      <b># self.disable_gravity()  # to prevent plug from falling</b>
      self._reset_object()
      self._move_gripper_to_grasp_pose(
          sim_steps=self.cfg_task.env.num_gripper_move_sim_steps
      )
      self.close_gripper(sim_steps=self.cfg_task.env.num_gripper_close_sim_steps)
      self.enable_gravity()

      # Get plug SDF in goal pose for SDF-based reward
      self.plug_goal_sdfs = algo_utils.get_plug_goal_sdfs(
          wp_plug_meshes=self.wp_plug_meshes,
          asset_indices=self.asset_indices,
          socket_pos=self.socket_pos,
          socket_quat=self.socket_quat,
          wp_device=self.wp_device,
      )

      self._reset_buffers()
</pre>

### Socket modifications

I modified the `_reset_socket` to three different versions, adding the following lines in different places (which I think should just read, and then set the friction parameters back to the same values):
```
for i, ptr in enumerate(self.env_ptrs):
    shape_props = self.gym.get_actor_rigid_shape_properties(ptr, self.socket_handles[i])
    self.gym.set_actor_rigid_shape_properties(ptr, self.socket_handles[i], shape_props)
```

To reproduce the issue of the peg passing through the socket, I used this version:
<pre>
    def _reset_socket(self):
        """Reset root state of socket."""

        # Randomize socket pos
        socket_noise_xy = 2 * (
            torch.rand((self.num_envs, 2), dtype=torch.float32, device=self.device)
            - 0.5
        )
        socket_noise_xy = socket_noise_xy @ torch.diag(
            torch.tensor(
                self.cfg_task.randomize.socket_pos_xy_noise,
                dtype=torch.float32,
                device=self.device,
            )
        )
        socket_noise_z = torch.zeros(
            (self.num_envs), dtype=torch.float32, device=self.device
        )
        socket_noise_z_mag = (
            self.cfg_task.randomize.socket_pos_z_noise_bounds[1]
            - self.cfg_task.randomize.socket_pos_z_noise_bounds[0]
        )
        socket_noise_z = (
            socket_noise_z_mag
[... 346 lines cut ...]

        # Stabilize plug
        self.plug_linvel[:, :] = 0.0
        self.plug_angvel[:, :] = 0.0

        # Set plug root state
        plug_actor_ids_sim = self.plug_actor_ids_sim.clone().to(dtype=torch.int32)
        self.gym.set_actor_root_state_tensor_indexed(
            self.sim,
            gymtorch.unwrap_tensor(self.root_state),
            gymtorch.unwrap_tensor(plug_actor_ids_sim),
            len(plug_actor_ids_sim),
        )

        # Simulate one step to apply changes
        self.simulate_and_refresh()
</pre>

### Remaining questions
- Is this solution to modifying the socket friction ok? Does it introduce some other bug that I'm not noticing? Why does the order matter in this case?
- Is there a way to modify the plug mass similarly?

Thanks for your help!

Patrick

bingjietang718, 2023-11-08:
Hi Patrick,

In general, gym setter functions (e.g., `self.gym.set_actor_root_state_tensor_indexed ` and `self.gym.set_actor_rigid_body_properties `) should only be called once in each simulation step (see [this post](https://forums.developer.nvidia.com/t/set-actor-rigid-shape-properties-doesnt-work-sometimes/212917) for reference). So what I would suggest is: 
- separate the pose reset of plug and socket (`reset_plug()` and `reset_socket()`) and the friction reset of plug and socket into different functions. 
- try to add `self.simulate_and_refresh()` within the for-loop instead of after the for-loop.

Best,
Bingjie

patricknaughton01, 2023-11-09:
Hi Bingjie,

I see, unfortunately, the simple fix of pushing `self.simulate_and_refresh()` into the for loop seems to cause some position updates to the plug to be ignored. Thanks for the pointer to the code, I'll check that out.

Patrick

closed 2023-11-11 by patricknaughton01