isaac-sim/IsaacGymEnvs #183: AttributeError: module 'numpy' has no attribute 'float' + Confusing FBX Package Error

ohernpaul, 2023-11-14:
OS: Ubuntu 20.04
Python: 3.8.0
IsaacGym: Preview 4

Initial error about FBX python library threw me off - this should be removed. Unless the user digs online to find the nvidia employee stating that the FBX library is only used for AMP demo and that this error should be ignored, they will spend a lot of time trying to get the fbx python sdk working, which will fail due to the incorrect version of python (fbx sdk needs python 10).

Actual Error: 
Numpy attribute for float is no longer existent for the version of numpy installed via IsaacGym (1.24.0). To fix this, I had to uninstall Numpy 1.24.0 and install the version with the last working usage of the float attribute (1.20.0).

```
AttributeError: module 'numpy' has no attribute 'float'.
`np.float` was a deprecated alias for the builtin `float`. To avoid this error in existing code, use `float` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.float64` here.
The aliases was originally deprecated in NumPy 1.20; for more details and guidance see the original release note at:
    https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
```

After doing this, training the cartpole demo worked. 

Requesting: 
Please remove or patch the FBX package error message, and please update the code calling the float attribute that was previously deprecated. 

Full Error Below:
```
(mlagents_python_3_8_0) testusr@testusr-MS-7D09:~/ML/IsaacGymEnvs-main/isaacgymenvs$ python train.py task=Cartpole
/home/testusr/anaconda3/envs/mlagents_python_3_8_0/lib/python3.8/site-packages/hydra/_internal/hydra.py:119: UserWarning: Future Hydra versions will no longer change working directory at job runtime by default.
See https://hydra.cc/docs/1.2/upgrades/1.1_to_1.2/changes_to_job_working_dir/ for more information.
  ret = run_job(
Importing module 'gym_38' (/home/testusr/ML/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_38.so)
Setting GYM_USD_PLUG_INFO_PATH to /home/testusr/ML/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
/home/testusr/anaconda3/envs/mlagents_python_3_8_0/lib/python3.8/site-packages/torch/utils/cpp_extension.py:28: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  from pkg_resources import packaging  # type: ignore[attr-defined]
/home/testusr/anaconda3/envs/mlagents_python_3_8_0/lib/python3.8/site-packages/pkg_resources/__init__.py:2871: DeprecationWarning: Deprecated call to `pkg_resources.declare_namespace('google')`.
Implementing implicit namespace packages (as specified in PEP 420) is preferred to `pkg_resources.declare_namespace`. See https://setuptools.pypa.io/en/latest/references/keywords.html#keyword-namespace-packages
  declare_namespace(pkg)
PyTorch version 2.1.0+cu121
Device count 1
/home/testusr/ML/isaacgym/python/isaacgym/_bindings/src/gymtorch
Using /home/testusr/.cache/torch_extensions/py38_cu121 as PyTorch extensions root...
Emitting ninja build file /home/testusr/.cache/torch_extensions/py38_cu121/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
Loading extension module gymtorch...
2023-11-13 17:33:24,741 - INFO - logger - logger initialized
<unknown>:3: DeprecationWarning: invalid escape sequence \*
Error: FBX library failed to load - importing FBX data will not succeed. Message: No module named 'fbx'
FBX tools must be installed from https://help.autodesk.com/view/FBX/2020/ENU/?guid=FBX_Developer_Help_scripting_with_python_fbx_installing_python_fbx_html
Error executing job with overrides: ['task=Cartpole']
Traceback (most recent call last):
  File "train.py", line 81, in launch_rlg_hydra
    from isaacgymenvs.utils.rlgames_utils import multi_gpu_get_rank
  File "/home/testusr/ML/IsaacGymEnvs-main/isaacgymenvs/utils/rlgames_utils.py", line 40, in <module>
    from isaacgymenvs.tasks import isaacgym_task_map
  File "/home/testusr/ML/IsaacGymEnvs-main/isaacgymenvs/tasks/__init__.py", line 56, in <module>
    from .industreal.industreal_task_pegs_insert import IndustRealTaskPegsInsert
  File "/home/testusr/ML/IsaacGymEnvs-main/isaacgymenvs/tasks/industreal/industreal_task_pegs_insert.py", line 46, in <module>
    from isaacgym import gymapi, gymtorch, torch_utils
  File "/home/testusr/ML/isaacgym/python/isaacgym/torch_utils.py", line 135, in <module>
    def get_axis_params(value, axis_idx, x_value=0., dtype=np.float, n_dims=3):
  File "/home/testusr/anaconda3/envs/mlagents_python_3_8_0/lib/python3.8/site-packages/numpy/__init__.py", line 305, in __getattr__
    raise AttributeError(__former_attrs__[attr])
AttributeError: module 'numpy' has no attribute 'float'.
`np.float` was a deprecated alias for the builtin `float`. To avoid this error in existing code, use `float` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.float64` here.
The aliases was originally deprecated in NumPy 1.20; for more details and guidance see the original release note at:
    https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.
```

leedokeun, 2024-05-03:
I have same problem with you. And I solve this install numpy <= 1.20.0

smmr0ago, 2024-11-09:
> I have same problem with you. And I solve this problem to install numpy <= 1.20.0

Thank you. Your answer solved the problem!