isaac-sim/IsaacGymEnvs #184: IndustRealSim and IndustRealLib

pt127, 2023-11-14:
hello, 
i really apriciate the work u did with IndustReal sofar. Do you maybe have a date in mind when the IndustRealLib tools to deploy the policy on real robots will be published? 
Also, does somebody know when and if IndustReal will be published on the OmniIsaacGym repository?
I would be very grateful for your answer :)

ynarang, 2023-11-27:
Hi @pt127,

Thank you for your feedback!

We have had the release version of IndustRealLib in a near-complete state for some time, and we aimed to release it shortly after IndustRealSim and IndustRealKit. However, we have had a few recent administrative and logistical changes on our team that have delayed the final push. We are now aiming for an early December release. Thank you for your patience!

We do not currently have any plans to integrate IndustRealSim with OmniIsaacGym, as we have very limited bandwidth at the moment. However, IndustRealSim is structured very similarly to [Factory for Isaac Gym](https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/tree/main/isaacgymenvs/tasks/factory), and we recently released [Factory for OmniIsaacGym](https://github.com/NVIDIA-Omniverse/OmniIsaacGymEnvs/tree/main/omniisaacgymenvs/tasks/factory) as well. The changes required to create an OmniIsaacGym version of IndustRealSim should be very similar to the diff between Factory for OmniIsaacGym and Factory for Isaac Gym. Please let us know if you do give it a try.

Best,
Yash

closed 2024-01-01 by kellyguo11

ynarang, 2024-02-06:
Hi @pt127,

This issue just came to mind. We released [IndustRealLib](https://github.com/NVLabs/industreallib) about a month ago. Feel free to give it a try.

Best,
Yash