isaac-sim/IsaacGymEnvs #185: No module named fbx

yqi19, 2023-11-20:
Hi, congratulations on your amazing work, when I was trying to install IsaacGymEnvs, I run into the issue that there is no module named fbx, and when I trying to solve it in https://aps.autodesk.com/developer/overview/fbx-sdk it seems that there is no wheel file at all. Then everything gets stuck. Does anyone know why?

Thank you!

```python
(eureka) yu@yu-G470:~/project/Eureka$ python test.py
Importing module 'gym_38' (/home/yu/project/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_38.so)
Setting GYM_USD_PLUG_INFO_PATH to /home/yu/project/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
/home/yu/anaconda3/envs/eureka/lib/python3.8/site-packages/torch/utils/cpp_extension.py:28: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  from pkg_resources import packaging  # type: ignore[attr-defined]
/home/yu/anaconda3/envs/eureka/lib/python3.8/site-packages/pkg_resources/__init__.py:2871: DeprecationWarning: Deprecated call to `pkg_resources.declare_namespace('mpl_toolkits')`.
Implementing implicit namespace packages (as specified in PEP 420) is preferred to `pkg_resources.declare_namespace`. See https://setuptools.pypa.io/en/latest/references/keywords.html#keyword-namespace-packages
  declare_namespace(pkg)
/home/yu/anaconda3/envs/eureka/lib/python3.8/site-packages/pkg_resources/__init__.py:2871: DeprecationWarning: Deprecated call to `pkg_resources.declare_namespace('google')`.
Implementing implicit namespace packages (as specified in PEP 420) is preferred to `pkg_resources.declare_namespace`. See https://setuptools.pypa.io/en/latest/references/keywords.html#keyword-namespace-packages
  declare_namespace(pkg)
PyTorch version 2.1.1+cu121
Device count 1
/home/yu/project/isaacgym/python/isaacgym/_bindings/src/gymtorch
Using /home/yu/.cache/torch_extensions/py38_cu121 as PyTorch extensions root...
Emitting ninja build file /home/yu/.cache/torch_extensions/py38_cu121/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
Loading extension module gymtorch...
2023-11-19 21:37:46,282 - INFO - logger - logger initialized
<unknown>:3: DeprecationWarning: invalid escape sequence \*
Error: FBX library failed to load - importing FBX data will not succeed. Message: No module named 'fbx'
FBX tools must be installed from https://help.autodesk.com/view/FBX/2020/ENU/?guid=FBX_Developer_Help_scripting_with_python_fbx_installing_python_fbx_html
Traceback (most recent call last):
  File "test.py", line 5, in <module>
    envs = isaacgymenvs.make(
  File "/home/yu/project/Eureka/isaacgymenvs/isaacgymenvs/__init__.py", line 29, in make
    env_path = cfg.env_path
AttributeError: 'NoneType' object has no attribute 'env_path'
```

locoxsoco, 2023-12-12:
I couldn't find the fbx sdk python wheel for python 3.8 either, so I created a new venv with python 3.10 where I could use the FBX Python SDK from the website you mentioned above to import fbx animations.

Jin0932, 2024-03-23:
> I couldn't find the fbx sdk python wheel for python 3.8 either, so I created a new venv with python 3.10 where I could use the FBX Python SDK from the website you mentioned above to import fbx animations.

Yes, I also meet this question, and how did you solve it?

(nvidiagym) jqwang@RL:~/jqwangFiles$ python test.py 
Importing module 'gym_38' (/home/jqwang16/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_38.so)
Setting GYM_USD_PLUG_INFO_PATH to /home/jqwang16/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
/home/jqwang16/miniconda3/envs/nvidiagym/lib/python3.8/site-packages/torch/cuda/__init__.py:141: UserWarning: CUDA initialization: The NVIDIA driver on your system is too old (found version 11040). Please update your GPU driver by downloading and installing a new version from the URL: http://www.nvidia.com/Download/index.aspx Alternatively, go to: https://pytorch.org to install a PyTorch version that has been compiled with your version of the CUDA driver. (Triggered internally at ../c10/cuda/CUDAFunctions.cpp:108.)
  return torch._C._cuda_getDeviceCount() > 0
No CUDA runtime is found, using CUDA_HOME='/usr'
PyTorch version 2.2.1+cu121
Device count 2
/home/jqwang16/isaacgym/python/isaacgym/_bindings/src/gymtorch
Using /home/jqwang16/.cache/torch_extensions/py38_cu121 as PyTorch extensions root...
Emitting ninja build file /home/jqwang16/.cache/torch_extensions/py38_cu121/gymtorch/build.ninja...
Building extension module gymtorch...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
ninja: no work to do.
Loading extension module gymtorch...
2024-03-23 11:22:30,767 - INFO - logger - logger initialized
<unknown>:3: DeprecationWarning: invalid escape sequence \*
Error: FBX library failed to load - importing FBX data will not succeed. Message: No module named 'fbx'
FBX tools must be installed from https://help.autodesk.com/view/FBX/2020/ENU/?guid=FBX_Developer_Help_scripting_with_python_fbx_installing_python_fbx_html
Traceback (most recent call last):
  File "test.py", line 7, in <module>
    envs = isaacgymenvs.make(
  File "/data/jqwang16Files/IsaacGymEnvs/isaacgymenvs/__init__.py", line 27, in make
    from isaacgymenvs.utils.rlgames_utils import get_rlgames_env_creator
  File "/data/jqwang16Files/IsaacGymEnvs/isaacgymenvs/utils/rlgames_utils.py", line 40, in <module>
    from isaacgymenvs.tasks import isaacgym_task_map
  File "/data/jqwang16Files/IsaacGymEnvs/isaacgymenvs/tasks/__init__.py", line 56, in <module>
    from .industreal.industreal_task_pegs_insert import IndustRealTaskPegsInsert
  File "/data/jqwang16Files/IsaacGymEnvs/isaacgymenvs/tasks/industreal/industreal_task_pegs_insert.py", line 46, in <module>
    from isaacgym import gymapi, gymtorch, torch_utils
  File "/home/jqwang16/isaacgym/python/isaacgym/torch_utils.py", line 135, in <module>
    def get_axis_params(value, axis_idx, x_value=0., dtype=np.float, n_dims=3):
  File "/home/jqwang16/miniconda3/envs/nvidiagym/lib/python3.8/site-packages/numpy/__init__.py", line 305, in __getattr__
    raise AttributeError(__former_attrs__[attr])
AttributeError: module 'numpy' has no attribute 'float'.
`np.float` was a deprecated alias for the builtin `float`. To avoid this error in existing code, use `float` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.float64` here.
The aliases was originally deprecated in NumPy 1.20; for more details and guidance see the original release note at:
    https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations

ckji, 2025-01-17:
> I couldn't find the fbx sdk python wheel for python 3.8 either, so I created a new venv with python 3.10 where I could use the FBX Python SDK from the website you mentioned above to import fbx animations.

but python3.10 is not suitable for IsaacGym, did you solve this problem?

xiaotangzh, 2025-03-22:
Choose "FBX Python SDK" to install rather than "Python Binding" or "SDKs and tools", you will find .whl file.

lianghu3124, 2025-07-07:
I have exactly the fbx lib issue and other python issue came along with it. After 4 hours trouble shooting, I finally fixed it. It turns out fbx is not a issue for me. you only need to degrade numpy version to 1.23.5(has to <1.24). Once the numpy issue is gone, fbx error also gone. I have no idea why is this.