isaac-sim/IsaacGymEnvs #187: Video capturing does not work

y8han, 2023-11-22:
Hi, thank you for offering such a wonderful tool! 
I have encountered a problem when trying to capture the video, and any help would be appreciated!
At first, I typed in the command `python3 train.py task=Ant test=True capture_video=True force_render=False`, and I got the following error 
> FileNotFoundError: [Errno 2] No such file or directory: 'Xvfb': 'Xvfb'. 

As suggested in another [issue](https://github.com/NVIDIA-Omniverse/IsaacGymEnvs/issues/101), I installed the following dependencies:
`sudo apt-get install xvfb xserver-xephyr tigervnc-standalone-server x11-utils gnumeric`.
However, I still got another error
>    h, w, pixfmt = frame_shape
ValueError: not enough values to unpack (expected 3, got 0)

Unlike his case, I am using Ubuntu 20.04 and Wayland has already been disabled. So even though I have rebooted the system, I still got the same error.

MickyFlowers, 2023-12-05:
I've met the same problem. Any ideas on how to solve this problem?

TAEM1N2, 2024-01-10:
I'm using ubuntu 20.04 and met a same problem.

paLeziart, 2024-06-03:
@MickyFlowers @TAEM1N2 Check #194 #126 for possible solutions