isaac-sim/IsaacGymEnvs #199: How to run the IsaacGym simulation environment in an Ubuntu container and display the visual training interface?

Lihongjie-123, 2024-01-03:
Hello, sir. I would like to consult an issue regarding running the IsaacGym simulation environment in a Docker container. My goal is to run the IsaacGym simulation environment in a Docker container and display the visual training results through a VNC desktop during training.

The environment I am using is as follows:

1. Ubuntu 20.04
2. Docker 20.10.8
3. GPU: Tesla T4 15360MiB
4. GPU Driver Version: 535.129.03
5. CUDA Version: 12.2
6. VNC: TigerVNC

When executing the joint_monkey.py example, I encountered the following error:
`root@bdd460d9bfa1:~/isaacgym/python/examples# python joint_monkey.py
Importing module 'gym_38' (/root/isaacgym/python/isaacgym/_bindings/linux-x86_64/gym_38.so)
Setting GYM_USD_PLUG_INFO_PATH to /root/isaacgym/python/isaacgym/_bindings/linux-x86_64/usd/plugInfo.json
pygame 2.5.2 (SDL 2.28.2, Python 3.8.10)
Hello from the pygame community. https://www.pygame.org/contribute.html
WARNING: Forcing CPU pipeline.
Not connected to PVD
+++ Using GPU PhysX
Physics Engine: PhysX
Physics Device: cuda:0
GPU Pipeline: disabled
WARNING: lavapipe is not a conformant vulkan implementation, testing use only.
Creating 36 environments
XIO:  fatal IO error 11 (Resource temporarily unavailable) on X server ":1"
      after 385 requests (385 known processed) with 0 events remaining.`

Tracking the code, I found that the error occurs when the program reaches the following line:
`gym.draw_viewer(viewer, sim, True)`

Could you please provide insights into the reason for this error? Any advice or solutions would be greatly appreciated!

Thank you very much!

Lihongjie-123, 2024-01-05:
However, when running the 'joint_monkey.py' example code in a container, the displayed visualization interface looks like this, with the interface being very blurry, and the characters on top are not visible.
[image]

Lihongjie-123, 2024-01-05:
Why is this happening? I am using a Tesla T4 GPU, and even after updating the GPU driver to version 510.108.03, I still encounter the same issue. The problem persists with GPU driver version 535.129.03 as well. Can you provide some advice or guidance?

Odin233, 2024-10-10:
Same problem. Have you solved this problem? ;_;

zhichenghe2008, 2024-11-20:
is there any issue if using --headless  mode to run isaacgym in docker container env?

zhichenghe2008, 2024-11-20:
I am also try to use isaacgym in docker conainer env , but when I run a simple isaacgym demo as flowws:
python
from isaacgym import gymapi
from isaacgym import gymutil
gym = gymapi.acquire_gym()
args = gymutil.parse_arguments(description="Example")
sim_params = gymapi.SimParams()
sim_params.up_axis = gymapi.UP_AXIS_Z
sim_params.dt = 1.0 / 60.0
sim = gym.create_sim(0, 0, gymapi.SIM_PHYSX, sim_params)

there still have issue as fllow:
Segmentation fault (core dumped)

MiangChen, 2024-11-21:
i have the same problem
GPU: 4070tis
NVIDIA Driver: 550.135
CUDA: 12.4
```
python joint_monkey.py
```

[image]

[image]

MiangChen, 2024-11-21:
try to downstrem your Driver and CUDA version may helps

GPU: 4070 Ti super(release 2024/2）
at first, I install  `NVIDIA-Linux-x86_64-550.135` and `cuda_12.4.1_550.54.15_linux`, and when I run `python 1080_balls_of_solitude.py `, segmentation fault occurs

then, I simply downgrade the version: `NVIDIA-Linux-x86_64-535.171.04` and `cuda_12.2.2_535.104.05_linux`, then, this bug disapper

so, i simply suggest that you can try to downgrade the NVIDIA Driver and Cuda version.

Earl-chen, 2025-01-13:
> However, when running the 'joint_monkey.py' example code in a container, the displayed visualization interface looks like this, with the interface being very blurry, and the characters on top are not visible. ![4e1b313c133af8d46339e0d4dd1ce89](https://private-user-images.githubusercontent.com/79447160/294434969-b5c6779d-ba5c-4e01-8397-914e10a4ff4f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY3NTUyNDgsIm5iZiI6MTczNjc1NDk0OCwicGF0aCI6Ii83OTQ0NzE2MC8yOTQ0MzQ5NjktYjVjNjc3OWQtYmE1Yy00ZTAxLTgzOTctOTE0ZTEwYTRmZjRmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTEzVDA3NTU0OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWM1M2YyYjNiNDNhZDM5NjExMjgzZmVhMmM3MWE3NWNjNjg4MjFiM2IzYjZlZmFiNGI2MzFiYjkxNWY3ZDljZmEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.Aw_p9ZvpqZbfwUT7cK4FEQwgIwZul60AmIEQq7sc7e8)

Same problem. Have you solved this problem? ;_;