isaac-sim/IsaacGymEnvs #201: How to load a textured mesh that can result in collision?

zjwzcx, 2024-01-04:
I found two ways to load mesh into Isaac Gym: 
**1.** self.gym.load_asset() -> self.gym.create_actor()
**2.** self.gym.add_triangle_mesh()

**What's the difference between them?** 

I found that the mesh loaded by load_asset() shows the texture while it cannot result in collision. The mesh loaded by add_triangle_mesh() can result in collision but shows no texture.

**Way 1:**
`
        urdf_name = "xxxx.urdf"
        asset = self.gym.load_asset(self.sim, asset_root, urdf_name, asset_options)
        pose = gymapi.Transform()
        ahandle = self.gym.create_actor(env_handle, asset, pose, None, env_index, 0)

`
**Way 2:**
`        mesh = o3d.io.read_triangle_mesh('active_reconstruction/objects/replica_v1/room_0/mesh_tri_tsfm.obj')
        mesh_vertices = np.asarray(mesh.vertices).astype(np.float32)
        mesh_triangles = np.asarray(mesh.triangles).astype(np.uint32)

        tm_params = gymapi.TriangleMeshParams()
        tm_params.nb_vertices = mesh_vertices.shape[0]
        tm_params.nb_triangles = mesh_triangles.shape[0]
        self.gym.add_triangle_mesh(self.sim, mesh_vertices.flatten(order='C'),
                                    mesh_triangles.flatten(order='C'),
                                    tm_params)`

CrossStyle, 2024-09-19:
Do your any solution now?

zjwzcx, 2024-09-20:
> Do your any solution now?

No.. I think Isaac Gym cannot support textured mesh. So I load meshes without texture using the second method (self.gym.add_triangle_mesh).

closed 2024-09-20 by zjwzcx